pbatard/rufus #2545: German ISO Win11 Bypass unsuccessfull

fokklz, 2024-08-19:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have an old computer on which I'd like to install Windows 11, even though it's not officially supported. When trying to create a bootable USB drive using Rufus, the process fails at around 60% completion.
After attempting this with several USB drives and different computers without success, I decided to try using the English ISO file, knowing that registry entries might be language-specific.
Using the English ISO worked perfectly. So it seems that, at least for the German version, this process doesn't work.
If this issue is known or intentional, I'd appreciate a prominent warning in the extra options section when creating the bootable drive.

Log
---
```
Rufus x64 v4.5.2180
Windows version: Windows 10 Pro x64 (Build 17134.1304)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
WARNING: Could not parse this system's SkuSiPolicy.p7b for additional revoked UEFI bootloaders
Found USB 3.0 device 'Innostor Innostor USB Device' (1F75:0917)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_DE-DE_DV9'
  Detected: Windows 11 ISO (Build 22621.2861)
  Size: 6.3 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_23H2_German_x64v2.iso (6.3 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): [0x80070032] (NB: This system was unable to provide an English error message)
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
[... 895 lines cut ...]
  Start Sector: 31455144, Attributes: 0x9000000000000000
Found USB 3.0 device 'Innostor Innostor USB Device' (1F75:0917)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {A5A8720E-9DB1-43E7-93E3-B5DD2D5EDAF9}
Max parts: 128, Start Offset: 17408, Usable = 16106093056 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {1E8CBD20-BB61-454C-9BE5-28D65136D59F}
  Size: 15.0 GB (16103968768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {B55F91C7-E034-4454-B65D-099DA3F7AD20}
  Size: 1 MB (1064448 bytes)
  Start Sector: 31455144, Attributes: 0x9000000000000000
0 devices found

```

JonnyTech, 2024-08-19:
>  If using an image, I clicked on the (✓) button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Please run the checksum test. It looks like you have a corrupted ISO file.

JonnyTech, 2024-08-19:
https://github.com/pbatard/rufus/wiki/FAQ#how-can-i-validate-that-a-windows-iso-is-a-genuine-retail-version

fokklz, 2024-08-19:
No, why should it, was freshly downloaded on three devices in 2 seperate networks. there is no way its corrupted 3 times.

Also there is `if applicable` for a reason.
but to satisfy you. Here you go. The one i tried on my laptop.

[image]

fokklz, 2024-08-19:
>After attempting this with several USB drives and different computers without success

No, how on earth should it corrupt on 3 devices seperatly. 
`if applicable`

[image]

fokklz, 2024-08-19:
I apologize for the misunderstanding and misrepresentation of your template. I didn’t read through the checklist as clearly as I should have. I was at work and didn’t have much time to spend on writing the bug report—sadly, bug reporting isn't part of my job description 

Thank you for your detailed response.

I set up a testing environment in Hyper-V with a dedicated hard disk specifically to flash the disk and test the process. Interestingly, using the same language settings for the device and the same images, I was able to successfully flash the disk this time around. This means that, unfortunately, I'm unable to reproduce the issue again.

However, its still weird to me how the issue occurred in the first place. Here are some details for context:

Device Environments: The first device was a laptop installed just a week ago (win11 default setup). It only has Windows Defender running in its default configuration—no special security software was installed or altered.

The second device was a freshly reset device with an older version of Windows 10. This was the device I originally intended to use the ISO on.

Behavior Differences: The issue occurred on both devices when using the German ISO. However, when I switched to the English ISO using the same instance of Rufus, everything worked fine. There were no changes made to the USB device or anything else between the attempts.

Given that the issue seems to resolve when switching ISOs, I'm struggling to understand how a security tool or any other factor could block one ISO and then allow another under the same conditions.

As a final note, the error occurred in my work environment with my laptop (which is not domain-controlled) and the USB drives used there. I'll try to reproduce the issue again this week and will provide the full error log from the first failed attempt and the successful flash after switching ISOs, if that helps. Unfortunately, I didn't save the log at the time since I didn't think about it.

fokklz, 2024-08-20:
I have no idea, run with the same laptop, same ISO, same USB Port and same USB. Just worked. While it yesterday did not on several devices. So confused.

Anyways sorry for wasting your time then.

[rufus.log](https://github.com/user-attachments/files/16670645/rufus.log)

closed 2024-08-20 by fokklz