pbatard/rufus #2546: Windows new changes to prevent Bypass

madmaxgrey, 2024-08-20:
im currently running windows 11 using rufus's bypass will i be affected by the new update once it's released from windows beta testing?

pbatard, 2024-08-20:
Nobody knows.

This question wil l**only** be answered the day Microsoft officially releases Windows 11 24H2. which is till months away.

See also:
* https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_Insider_builds_NOT_SUPPORTED
* https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_work_around_the_new_hardware_requirements_limitations_of_Windows_11_24H2

closed 2024-08-20 by pbatard

thecatontheceiling, 2024-08-20:
If your hardware doesn't support Windows 11 you're exempt from Feature Updates entirely.

You have to be running IoT Enterprise edition edition in order to upgrade to 24H2 without meeting requirements.

thecatontheceiling, 2024-08-20:
@pbatard Not sure if I should make a separate issue for this or not but you **really** need to add some sort of warning that using the requirements bypass will completely prevent you from installing feature updates and will eventually lead to the user being on an EOL version of Windows (which means - no security updates).

pbatard, 2024-08-20:
> will completely prevent you from installing feature updates

I'm currently installing feature updates just fine on unsupported systems, by creating a drive with Rufus and running `setup.exe`.

So "completely prevent" is nonfactual. I've personally gone through 2 feature updates this way and did not have to reinstall Windows from scratch to bump these systems to the latest major refresh.

> and will eventually lead to the user being on an EOL version of Windows

Which is pure speculation at this stage. I don't warn about speculative. Else, I might as well _speculate_ that someone will find a way to work around whatever new limitations Microsoft *might* introduce in the future, thereby making the need for a **pre-emptive** warning unwarranted.

Again, as long as 24H2 has not been formally released, **NOTHING** needs to be done.

madmaxgrey, 2024-08-20:
> Nobody knows.
> 
> This question wil l**only** be answered the day Microsoft officially releases Windows 11 24H2. which is still months away.
> 
> See also:
> 
>     * https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_Insider_builds_NOT_SUPPORTED
> 
>     * https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_work_around_the_new_hardware_requirements_limitations_of_Windows_11_24H2

Thank you for the reply. I guess we'll only find out what happens when these updates are pushed to everyone. For now, I'll prepare a backup plan in case this happens. and probably leaves us locked out of our systems because of a feature update

thecatontheceiling, 2024-08-20:
> I'm currently installing feature updates just fine on unsupported systems, by creating a drive with Rufus and running `setup.exe`.
>
> So "completely prevent" is nonfactual. I've personally gone [...]

My wording was a bit off, but being cursed to use Rufus for every single feature update is still not ideal. 

People expect that the option in Rufus to bypass requirements wouldn't cause some updates to fail to install but of course that isn't true. 

You can't expect people to know that they need to download a new ISO and Rufus every time they want to install feature updates when there's nothing telling them that bypassing requirements causes Windows Update to complain. 

All I'm asking for is a simple disclaimer. A tooltip or something.

Most people using Rufus don't even know that setup.exe in the root of the ISO file is usable for upgrades. 

I've seen someone else give this exact same suggestion already but there was no response given by you

pbatard, 2024-08-20:
> People expect that the option in Rufus to bypass requirements wouldn't cause some updates to fail to install

I'm sorry but I'm going to counter that people who willingly apply a bypass are foolish if they expect that it will be business as normal.

I don't make the rules of the game. Microsoft does. If you **choose** to skirt Microsoft's rules, then you also tacitly imply that you are going to do your homework and not require someone, to put in black on white, that, yeah, if Microsoft introduced limitations and system requirements for reasons, it's only a smoke screen and you really can run Windows, in a manner that the manufacturer of the OS does not want you to run Windows, as if your system did fit the system requirements.

I am **not** going to start plastering screens all over the place (that people don't read anyway, see the _"hey, you should try DD mode if you get into an issue in ISO mode"_ that most people who run into trouble happen to ignore) just because people do not want to do their homework on account that they explicitly **chose** to go against the OS manufacturer's explicit recommendations.

Besides, placing a warning, for something that, at best, will only *suddenly* become relevant months after some installed their system (since there's only one refresh per year) **IS UTTERLY POINTLESS** because nobody will remember that they saw a vague warning at one time... *months prior*.

That's just not how warning works. And that's a lesson that's only taught to you, in actual black on white, when you actually produce warnings in software you publish at large and see the user feedback.

It is not Rufus' job to hold your hand. You are installing an OS. You are not playing Minecraft (and even Minecraft, as far as I can tell, doesn't really hold your hand, and does expect you to do your homework).

pineapple63, 2024-08-22:
I have come across computers where the hardware requirements had been bypassed, and the user was unaware that updates may not install

pbatard, 2024-08-22:
> This is something you must do (...) it's your job to tell people (...)

I am not going to reiterate the points I have already made about:
1. Rufus not being aimed at letting everybody and their grandmother install an OS as if it was a trivial operation that they will be guided through. It was never ever designed that way, and never will.
2. People having to do their homework, especially if they use what is called a **bypass**, with all the implications that the name implies.

Also, because it doesn't go your way, You are also assuming that I am misreading the user feedback, that I am actually actively seeking (i.e. it's not just people contacting me with an issue) and that I am misinterpreting social media post after social media post I see about people who used ISO mode and who 100% disregarded the ISO vs DD warning.

Well, if you do not want to accept a reality because that reality is not the one that you would like, I'm not going to go into yet another lengthy explanation of why I both see very little point to adding the extra warning **some** people are adamant Rufus should display about a mode of installation of Windows **that is CLEARLY NOT supported by the OS manufacturer and will therefore have likely detrimental consequences**, as well as being told that I should no longer have final say in the software I design.

So, congratulations, I will lock this thread, and tell you that, if you are unhappy about what the choices that the main designer of an Open Source application makes, you can just go and fork your own version.

When people use bypass tricks, they **should** expect the system to "blow up" (if you want to call the *current* requirement of having to do extra work to install Feature Update "blowing up"... Hyperbole much?). And if they don't, then I'd rather they learn that lesson the hard way than continue to think that OS manufacturers place system requirements, that actually serve no purpose, just to annoy people...