pbatard/rufus #2547: Website is completely non-functional

adnanwahab, 2024-08-21:
Hi, not sure if the website at rufus.ie is hacked but it has extremely blocking ads to something called browser which appears to be adware / spyware ?

Is there an alternate place to download?

thecatontheceiling, 2024-08-21:
Introduce ad-blockers into your life

google "ublock origin"

thecatontheceiling, 2024-08-21:
https://github.com/pbatard/rufus/issues/2543

pbatard, 2024-08-21:
Hmmm, the ads should not be blocking. At least I'm pretty sure I explicitly tell Google, in the ads settings, that full screen ads are not allowed (but then again, when it comes to ads, Google will always be trying to get away with stuff you don't want if it can help them maximize their revenue streams).

Do you (or anyone) have a screenshot of what you actually see, so that I get an idea of how intrusive these latest ads might be (since I don't personally see them)?

And as @thecatontheceiling reported, because it is essentially a problem with ad abuse, if you see a deceptive ad, it needs to be reported to Google so that they filter it out at the source.

> Is there an alternate place to download?

https://rufus.ie/downloads/ (which might still display ads) or https://github.com/pbatard/rufus/releases.

closed 2024-08-22 by adnanwahab