pbatard/rufus #2552: Bring Back Support For Vista And XP!

Trancer03, 2024-08-27:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT reoved any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
It needs **ldlinux.sys**, I've found it on **http://rufus.ie/files/syslinux-6.04/ldlinux.sys**, but don't know where to place it!
Log
---
```
Rufus version: 2.9.934
Windows version: Windows Vista SP2 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Using autorun.inf label for drive F: 'Peppermint Ten'
Found USB device 'Multiple Card Reader' (????:????)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 246, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x33445FD8
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 1.9 GB (2022703104 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'PeppermintOS'
  Size: 1592279040 bytes
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: PeppermintOS-Debian-64.iso
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
Checking for Rufus updates...
Checking release channel...
Checking beta channel...
Updates: Unable to access version data
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
<FULL LOG>
```

JonnyTech, 2024-08-27:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

Trancer03, 2024-08-27:
Can I still do something with this file?
I will give you as much money as you want! Look at my profile.

Trancer03, 2024-08-27:
I guess I'm running **portable** version. Is there a link to the **standard** version that I'm running?

thecatontheceiling, 2024-08-27:
Delete this issue and avoid embarassing yourself further.

JonnyTech, 2024-08-27:
> I guess I'm running portable version. Is there a link to the standard version that I'm running?

https://github.com/pbatard/rufus/wiki/FAQ#user-content-How_comes_the_portable_and_regular_versions_are_binary_identical

closed 2024-08-27 by pbatard

Trancer03, 2024-09-12:
> Delete this issue and avoid embarassing yourself further.

I'm not embarrassed at all. No question is a stupid question!

thecatontheceiling, 2024-09-12:
"People who haven't pooped in 2024 yet, why are you still holding on to last years shit?"

pbatard, 2024-09-12:
@thecatontheceiling, I have deleted your last comment, because I don't want this discussion to become a back and forth of disparaging comments, which it is likely to devolve into if you start going the route you were going. I appreciate your contributions, but please refrain from using inflammatory one-liners, thank you.

I am also locking this discussion since it has been answered and doesn't warrant further communication.