pbatard/rufus #2555: how to tell what verison of rufus was used to create a usb key. and how to update the usb key with latest version?

asdffdsazqqq, 2024-08-29:
﻿Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
hello, thanks for rufus, it runs perfectly.  please i cannot figure out the following:
1. how to know what version of rufus and uefi-ntfs is in use on a usb key?
all i see is README.txt, which does not contain version information. i also tried to use look at the properties, using windows explorer, of the .efi images, but no version numbers?
2. how to update a usb key with the latest version of rufus and uefi-ntfs. as i can do with ventoy?
again, everything is working great, thanks much for your work!

thecatontheceiling, 2024-08-29:
By downloading the latest version of Rufus and re-flashing the ISO

asdffdsazqqq, 2024-08-29:
thanks much for the quick reply. 
i am looking to update an existing usb key without re-formatting the usb key or affecting all the custom work i have done on it.

is there any documentation about that, is there a button or function to update?
rufus does not seem to be able to scan an existing rufus install, no option to update just the rufus and uefi:ntfs parts of the usb.

for example, ventoy will scan a usb key, to see if an older version of ventoy was installed and then offer a simple update.

pbatard, 2024-08-29:
I'm going to have to ask why you think you need that information and why you think you need to update a drive unless it isn't working (but you are stating that everything runs fine).

Rufus does not create version data on a USB drive because (unlike what is the case for Ventoy or other utilities) it does not inject content into the original content. So either a drive works or it doesn't, and if it doesn't, and you think this is due to something that was fixed in a more recent version of Rufus (like partition alignment data, improved support for some Linux distros), you do **NOT** attempt to hack a fix on your own, which is almost guaranteed to lead to more issues, but you just recreate the drive using the latest version of Rufus. And if you are unsure if a new version was released since you created your media, just look at the date of the last release from the web site, as you are expected to have *some* idea of when you created your drive, and whether it might be before or after the most recent Rufus release.

There is *no* updating of an existing usb key. This is not a move of operation that will ever be supported in Rufus. Instead, your mode of operation should be:

1. If you have an existing USB key, and think that it contains the OS you want to install, try it and see if it boots and installs your OS.
2. If it works, you're done, and there's absolutely **nothing** that you are going to gain from trying to somehow "update" it with the latest version of Rufus. It will not boot or install the OS any better than it did then. If it doesn't work, then you should download the latest version of Rufus, recreate media, and see how that goes.
3. If it still doesn't work, and you are sure that you are using the latest version, then you should have a look at the FAQ, and if you cannot find an answer to your issue, report the problem here.

That's how you're supposed to proceed.

Also, please don't cross post.

closed 2024-08-29 by pbatard

asdffdsazqqq, 2024-08-29:
thanks much, very sorry about the cross post. this question is about rufus and that other post was about uefi:ntfs.
so i am confused about where to post? 

well, the current version of uefi:ntfs that was installed by rufus is `2.2`, which is three years old.

so i assumed, same as with any software, there are bug fixes and other such improvements in the past three years?
newer version of .efi files. as it is a usb key, it is plugged into many systems, not just one machine and i see in the changelog, about better handling of certain hardware machines.

just one example, i checked the change log for "Version 4.5 (2024.05.22)"
"Update [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) to latest (now always uses the [ntfs-3g](https://github.com/pbatard/ntfs-3g) driver, rather than the [buggy AMI NTFS one](https://github.com/pbatard/AmiNtfsBug))"

asdffdsazqqq, 2024-08-30:
@pbatard, you software is excellent, you are the expert, thanks very much for taking the time to respond to me.
i just tried to donate, but i see that you are not into that.

in the end, i will never understand that logic of not updating software, efi images and so on.

can please close this topic.