pbatard/rufus #2561: Add 64-bit QWORD-values to Windows 11 install TMP2.0 bypass option

Lumipyry, 2024-09-13:
﻿Thank You for this amazing program. I have a suggestion here, but first the problem:

While using Rufus to install Windows 11 to an unsupported machine by using the TMP2.0 bypass option, I noticed that install procedure made many loops at the phase when install was fetching updates. This phase started many times again.

I have installed Windows 11 several times during last few days. The problem of loop was repeatable.

Adding a new key to REGEDIT with 64-bit QWORD-values to:
Computer/HKEY_LOCAL_MACHINE/SYSTEM/Setup seems to solve the problem.

There were already 32-bit DWORD-values in a key named "LabConfig"  - supposedly made by Rufus?
I named the 64-bit values with similar names and values as the 32-bit values. 

So,
If you want to make Rufus even more perfect - you could add a similar key for 64-bit QWORD values as the 32-bit DWORD-values.

Yours,
Sami Saarinen

Lumipyry, 2024-09-14:
Please delete this Issue.
I was too hasty in writing this solution - it did not help after all.
I could not install Windows 11 even with 64-bit QWORD-values to this particular machine (iMac),
so the problem is somewhere else... (Althought I have been able to install this OS to this  machine before)

pbatard, 2024-09-14:
Yeah, I was about to reply that your request made absolutely no sense, because there's no way what you describe could have an impact, and you have to be exceedingly careful not to assume that a behavioural change that you saw after trying things in desperation means that one of these things you tried was the cause of the change, and that it can be generalized.

I'm going to close this issue then.

closed 2024-09-14 by pbatard