pbatard/rufus #2562: WIN11 blue screen crash

duaneellissd, 2024-09-18:
I have created a bootable UEFI image using the "BMP-TOOLS" from the Yocto KIRKSTONE 
I create this with the "bmptool" 
Under Linux, I would write this directly to the BlockDevice (aka: My usb thumb drive).

However - Linux machine I am using today is a VM. and I cannot plug a USB drive into this.
So - Step 1 I use BMPTOOLS to create a DISK IMAGE...
Step 2: I copy the DISK IMAGE to my Windows 11 laptop.
Step 3: I run RUFUS 4.4 select my image file and the USB DRIVE
Step 4: It asks: "are you sure" do you want to overwrite the disk loose all data etc.
Step 5: I answer(click) YES
Step 6: About 1 or 2 seconds later "BANG" - Blue Screen
The blue screen lasts a few seconds and reboots
When it reboots it tells me error code 0xC0000001
totally repeatable...

I have used RUFUS with other things - ie: to create a RASPBERRY PI bootable sd card..
The image file I am writing is about 2.4GIG

duaneellissd, 2024-09-18:
Additional information - using GITBASH on windows, I can do this: (NOTE: I have to run gitbash as admin)

dd if=MYFILE of=/dev/sdc BS=64k

and that does not crash when I do that

duaneellissd, 2024-09-18:
Hmm - about the LOG suggestion:

what and where would I get a log ... Its less then 1 second and it crashes

pbatard, 2024-09-18:
> what and where would I get a log ... Its less then 1 second and it crashes

As explained above you could provide the log before the crash, which would provide us with some of the data we need to try to investigate the issue.

Plus you also seriously expect me to do anything about your issue **WITHOUT** sharing the actual image you used? If it's totally repeatable, then you **totally** need to share your image. I have zero time to try to use some external tool in the hope that I might be able to figure out your options and possibly be able to replicate your issue. Unless you upload the actual image you use somewhere, I will simply close this issue.

pbatard, 2024-09-25:
No image to test with &rarr; Closed.

closed 2024-09-25 by pbatard