pbatard/rufus #2563: Create an ISO from RUFUS (using the benefits of RUFUS' Windows 11 Bypass)

JSchimmelpfennig, 2024-09-26:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

Hello rufus lovers :)
I'm using RUFUS for years, especially for the Windows Installation bypasses. Now I would like to install a Windows 11 VM - but I couldn't find an option in RUFUS to write to an ISO instead of a USB device. I know, the ISOs are usually the input and in this case the Windows 11 ISO is the input, but I would like to benefit of RUFUS bypasses to create a Windows 11 bypasses ISO. Furthermore, this would be useful to use this ISO on my ventoy USB stick as well for other, physical computers.

Is that already possible and I just didn't find the option?

Thank you :)

pbatard, 2024-09-26:
Answered in https://github.com/pbatard/rufus/wiki/FAQ#user-content-Do_you_plan_to_add_USB__ISO_creation_support

Just create a USB or a VHD and then use one of the many tools that can create an ISO.

closed 2024-09-26 by pbatard

nikkdrive18, 2024-10-18:
Can you do ISO to ISO please
I tried doing USB to ISO with Imgburn the ISO file boots in BIOS but not in UEFI.
I use Ventoy to keep all ISO Files. Thats why i need ISO file.
Can you please add "Save as iso" in the USB selection menu.

Also can you add links to where we can find more details about how to apply the W11 mods to -[https://github.com/pbatard/rufus/wiki/FAQ#could-you-please-provide-a-way-to-bypass-windows-restriction-x-just-like-you-did-for-tpm-and-the-other-stuff](url)