pbatard/rufus #2564: Rufus' newer versions can't create bootable USB for older PCs even if "BIOS compatibility" is checked

BerBerOnGithub, 2024-09-27:
﻿

Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm trying to create a FreeDOS bootable USB for an older PC, but no matter what USB drive I use, it dosen't show up on the old PC. It creates a second partition, as seen in the log, but the PC still dosen't detect the USB drive, even with "USB Emulation" enabled in the BIOS. An older version of Rufus works, but I don't want to deal with the inconvienience of having 2 versions of Rufus at the same time.

Log
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Education x64 (Build 22635.4225)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x040E (hu-HU)
Will use default UI locale 0x040E
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'hu-HU'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 3.7 GB (4008161280 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Could not open GUID volume '\\?\Volume{f5bda224-7ce2-11ef-a506-005056c00008}': [0x00000021] The process cannot access the file because another process has locked a portion of the file.
Opened \\.\PhysicalDrive3 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 131072, size: 3.7 GB)
● Creating BIOS Compatibility Partition (offset: 4009195008, size: 31.5 KB)
Waiting for logical drive to reappear...
Formatting to FAT (using IFS)
Using cluster size: 65536 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive3 for shared write access
Writing Master Boot Record...
Partition is already FAT16 LBA...
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume \\?\Volume{f5bda406-7ce2-11ef-a506-005056c00008}\
Opened \\?\Volume{f5bda406-7ce2-11ef-a506-005056c00008} for exclusive write access
Writing Partition Boot Record...
Using FreeDOS FAT16 partition boot record
Confirmed new volume has a FAT16 boot sector
Successfully remounted \\?\Volume{f5bda406-7ce2-11ef-a506-005056c00008}\ as F:
Copying DOS files...
Successfully wrote 'F:\COMMAND.COM' (85480 bytes)
Successfully wrote 'F:\KERNEL.SYS' (46256 bytes)
Successfully wrote 'F:\LOCALE\DISPLAY.EXE' (3657 bytes)
Successfully wrote 'F:\LOCALE\KEYB.EXE' (11446 bytes)
Successfully wrote 'F:\LOCALE\MODE.COM' (16244 bytes)
Successfully wrote 'F:\LOCALE\KEYBOARD.SYS' (40360 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD2.SYS' (29750 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD3.SYS' (32178 bytes)
Successfully wrote 'F:\LOCALE\KEYBRD4.SYS' (13105 bytes)
Successfully wrote 'F:\LOCALE\EGA.CPX' (6464 bytes)
Successfully wrote 'F:\LOCALE\EGA2.CPX' (7059 bytes)
Successfully wrote 'F:\LOCALE\EGA3.CPX' (5469 bytes)
Successfully wrote 'F:\LOCALE\EGA4.CPX' (4431 bytes)
Successfully wrote 'F:\LOCALE\EGA5.CPX' (7217 bytes)
Successfully wrote 'F:\LOCALE\EGA6.CPX' (7409 bytes)
Successfully wrote 'F:\LOCALE\EGA7.CPX' (5387 bytes)
Successfully wrote 'F:\LOCALE\EGA8.CPX' (6973 bytes)
Successfully wrote 'F:\LOCALE\EGA9.CPX' (5785 bytes)
Successfully wrote 'F:\LOCALE\EGA10.CPX' (5543 bytes)
Successfully wrote 'F:\LOCALE\EGA11.CPX' (7228 bytes)
Successfully wrote 'F:\LOCALE\EGA12.CPX' (8119 bytes)
Successfully wrote 'F:\LOCALE\EGA13.CPX' (6281 bytes)
Successfully wrote 'F:\LOCALE\EGA14.CPX' (7758 bytes)
Successfully wrote 'F:\LOCALE\EGA15.CPX' (6458 bytes)
Successfully wrote 'F:\LOCALE\EGA16.CPX' (7793 bytes)
Successfully wrote 'F:\LOCALE\EGA17.CPX' (8929 bytes)
Successfully wrote 'F:\LOCALE\EGA18.CPX' (5158 bytes)
Windows KBID 0x0000040e
Will use DOS keyboard 'hu' [Hungarian]
Will use codepage 852 [Central-European]
Successfully wrote 'CONFIG.SYS'
Successfully wrote 'AUTOEXEC.BAT'
Finalizing, please wait...

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01151F4C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Detected File System: FAT16
  Size: 3.7 GB (4009033728 bytes)
  Start Sector: 256, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Detected File System: (Unrecognized)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 7830459, Boot: No

```

pbatard, 2024-09-27:
Thanks for the report. By the looks of it, we may have introduced a regression where Rufus doesn't write the FreeDOS MBR, but writes the Windows specific Rufus MBR to the USB drive, in which case the drive may indeed not boot properly to DOS.

BerBerOnGithub, 2024-09-27:
Note: I tried it with multiple variations (Linux Distributions, Windows installers, etc.) and none of them make it appear unless I use an older version of Rufus.

pbatard, 2024-09-27:
Then that's something else, as Rufus is downloaded close to 3 million times each month, and if version 4.5, that was released more than 4 months ago, had the kind of universal problem you describe, you'd see thousands of complaints reporting a similar problem to yours in your issue tracker.

Considering that you do get `The process cannot access the file because another process has locked a portion of the file.` during drive creation, and that I've seen this kind of issues before, my guess is that you may have a security solution where older versions of Rufus have been whitelisted but newer versions haven't, that prevents proper access to the drive, and therefore corrupts the drive creation.

At any rate, if none of the drives created by Rufus work, be it for Windows, Linux or DOS, I can guarantee that your issue is purely environmental, because, again, many, many people would have reported something similar by now, and you are the very first, in 4 months of the v4.5 release, to report complete failure to create anything bootable.

pbatard, 2024-09-29:
For the record, I just validated that the use of Rufus MBR is a non issue and that a FreeDOS bootable USB created by Rufus has no issue booting FreeDOS on a real-life system.

Therefore, I have to reiterate that your issue is purely environmental and does not come from Rufus itself (especially as the FreeDOS creation code has not changed for a while.

pbatard, 2024-10-05:
No update &rarr; Closed.

closed 2024-10-05 by pbatard