pbatard/rufus #2567: Won't work with new Win11 24H2 ISO

mikiem2, 2024-10-01:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Doesn't work to setup installation USB stick for just released Win11 24H2 english us SHA256: b56b911bf18a2ceaeb3904d87e7c770bdf92d3099599d61ac2497b91bf190b11 [in case new ISO posted]. NOT as issue for me -- created the USB stick using Insider ISO instead, where Rufus worked flawlessly with same hardware etc., but wanted to notify.

Log
---
```
Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 11 Pro x64 (Build 22631.4249)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4100)
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
2 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 973, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00D6F655
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 7.5 GB (8007974912 bytes)
  Start Sector: 2048, Boot: No
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1926, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4862D9D2-119A-4C3A-971A-ED876F782B26}
Max parts: 128, Start Offset: 17408, Usable = 15846046208 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {775AC309-292A-4AD1-A09D-B3768962DBCF}
  Size: 14.8 GB (15844999168 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 26100.1742)
  Size: 5.4 GB (Projected)
  Has a >4GB file
[... 155 lines cut ...]

Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4100)
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1926, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {B0DF8142-BC77-445B-A4EC-B7CA3CF98C81}
Max parts: 128, Start Offset: 17408, Usable = 15846046208 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {C7685B9E-287E-4FCA-AC19-C8CA49CED8F8}
  Size: 14.8 GB (15843917824 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {C2954EEC-F7BA-4ACA-9A33-22730275CCAB}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30947238, Attributes: 0x9000000000000000

```

pbatard, 2024-10-02:
```
Write error [0x00000005] Access is denied.
```

Issues is 100% with your environment, and not Rufus.
See https://github.com/pbatard/rufus/wiki/FAQ#user-content-Error_0x00000005_Access_is_denied

I tested the Windows 11 24H2 ISOs a few hours ago and saw no issue.

closed 2024-10-02 by pbatard

Dhyfer1, 2024-10-02:
@pbatard I have the same problem. I have downloaded the ISO for 24H2 from Rufus and also from the official Microsoft site. The ISO was mounted with the option to remove the ram requirement, safe boot and tpm. There are no errors in the whole process, and when Rufus finishes I just reboot my PC to boot with the USB. But the USB does not boot and Windows starts normally in version 23H2

pbatard, 2024-10-02:
>  I have the same problem.

Unless you get `Write error [0x00000005] Access is denied.` when you create the drive, then you **definitely** do not. And as you stated yourself, you did not get errors in the whole process, ergo, you do **NOT** have the same problem.

> But the USB does not boot

UEFI/BIOS configuration issue, or incompatibility on your side. Especially, I'm pretty sure Windows has stopped to support BIOS/Legacy boot, so if you try to boot in legacy mode, that's not going to work.

Does the system you are trying to boot on support UEFI boot?

mikiem2, 2024-10-02:
> I'm pretty sure Windows has stopped to support BIOS/Legacy boot

Just purely FWIW, I have a non hardware compliant V/Box VM running Win11 24H2 using legacy boot without any problems, in case that helps anything. My concern was that ***maybe*** Microsoft changed something, e.g., the fake appraiserres.dll will not allow upgrading on non-compliant hardware, and the "/product server" reportedly stopped working on canary builds.

pbatard, 2024-10-02:
And just for the record, I tested a 24H2 clean install on a fully unsupported system (an old Dell Optiplex 390, that has no support for Secure Boot, let alone any kind of TPM) and, as opposed to all the reports I got of _"24H2 is going to break the bypasses"_, was surprised that it worked without much of an issue (I had to insist a bit on the new partition screen Microsoft appear to have introduced, which seems to be rather unresponsive at times, but aside from that, everything else run without a hitch).

Now, the only thing that I found doesn't work was in-place upgrade (there again on a fairly old Core i7 5930K based system without TPM), but I found that this could be solved by issuing the following (that I picked from [Win 11 Boot And Upgrade FiX KiT](https://forums.mydigitallife.net/threads/win-11-boot-and-upgrade-fix-kit-v5-0-released.83724/)) before running `setup.exe` on a drive created by Rufus with the bypass (so that `appraiseres.dll` gets replaced by a dummy zero byte version):
```
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers /f 2>NUL
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\Shared /f 2>NUL
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\TargetVersionUpgradeExperienceIndicators /f 2>NUL
reg.exe add HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\HwReqChk /f /v HwReqChkVars /t REG_MULTI_SZ /s , /d "SQ_TpmVersion=2,"
reg.exe add HKLM\SYSTEM\Setup\MoSetup /f /v AllowUpgradesWithUnsupportedTPMOrCPU /t REG_DWORD /d 1
```

So, so far, I'm not seeing much of a hurdle to running a 24H2 clean install or in-place upgrade on non officially supported system.

But of course, this is just my current experience with only 2 systems tested.

Dhyfer1, 2024-10-02:
@pbatard Have you already tried on a PC that does not have the POPCNT instruction or what is the same SSE4.2? In this case Windows 11 24H2 mounted with Rufus does not work.

As I saw that I could not boot the USB from the BIOS, I ran the setup, which shows me this 

[image]

Search the internet about the first requirement on the list and I found this [article](https://www.wincustomize.com/article/526097/disturbing-news-hardware-requirement-for-update-24h2), which says that for AMD processors SSE4A is required (SSE4.2 for Intel processors). So, according to CPU-Z my PC does have that instruction, as the image shows

[image]

Then I don't understand why I couldn't boot the USB mounted with the new version of Windows

pbatard, 2024-10-02:
> Have you already tried on a PC that does not have the POPCNT instruction or what is the same SSE4.2?

That's like asking if I tried on a 486DX2. I'm not Microsoft. I don't have an array of old machines to test with. And even if I did, I wouldn't bother testing on hardware that I consider too obsolete.

>  I ran the setup, which shows me this

Have you applied the registry commands I posted above? Coz I'm pretty sure you didn't.

But again, there's only so far I am going to go trying to support very old hardware. My testing on what I consider to be old hardware was successful enough to show that, as long as you're not being unreasonable in trying to run Windows 11 on PCs that was  designed to run XP or Vista, you should be fine. So it's going to be up to you to figure stuff out there.

mikiem2, 2024-10-02:
I apologize for causing any fuss -- my intent was only to point out a possible, potential issue in case it was of interest or use to others, 

> There is probably some software on your computer which is taking a disliking to specific ISOs

I tried, and failed, using a copy of Win10 & Win11 24H2 that I use exclusively for testing purposes -- both have *Only* 2 apps installed, 7Zip & Firefox.