pbatard/rufus #2568: W11 24H2 26100.1882 not installable on W11 23H2 22631.4249 PC

100PIER, 2024-10-02:
﻿

Checklist
---------
- [ x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<The FUFUS created W11 24H2 64bit FR  USB key does refuse W11 installation on my PC running W11 23H2>

Log
---
```
<
![Screenshot 2024-10-01 200238](https://github.com/user-attachments/assets/1dedfe3f-a538-412c-b89c-dc02ef5c67c8)
 
[rufus_W1124H2x64FR_01oct24.log](https://github.com/user-attachments/files/17221219/rufus_W1124H2x64FR_01oct24.log)
FULL LOG>
```

none2003, 2024-10-02:
Similar problem, used rufus 4.5 and chosen "Remove requirement for 4GB+ RAM, Secure Boot and TPM 2.0", but the installation program promotes "TPM 2.0 is mandatory" thing.

pbatard, 2024-10-02:
For in-place upgrades, you need to run the following from an elevated command prompt before running `setup.exe`:

```bat
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers /f 2>NUL
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\Shared /f 2>NUL
reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\TargetVersionUpgradeExperienceIndicators /f 2>NUL
reg.exe add HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\HwReqChk /f /v HwReqChkVars /t REG_MULTI_SZ /s , /d "SQ_SecureBootCapable=TRUE,SQ_SecureBootEnabled=TRUE,SQ_TpmVersion=2,SQ_RamMB=8192,"
reg.exe add HKLM\SYSTEM\Setup\MoSetup /f /v AllowUpgradesWithUnsupportedTPMOrCPU /t REG_DWORD /d 1
```

I'll add this batch script to the created media in a future version of Rufus.

JimBim89, 2024-10-02:
> For in-place upgrades, you need to run the following from an elevated command prompt before running `setup.exe` (credits to [Win 11 Boot And Upgrade FiX KiT](https://forums.mydigitallife.net/threads/win-11-boot-and-upgrade-fix-kit-v5-0-released.83724/)):
> 
> ```batchfile
> reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers /f 2>NUL
> reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\Shared /f 2>NUL
> reg.exe delete HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\TargetVersionUpgradeExperienceIndicators /f 2>NUL
> reg.exe add HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers\HwReqChk /f /v HwReqChkVars /t REG_MULTI_SZ /s , /d "SQ_SecureBootCapable=TRUE,SQ_SecureBootEnabled=TRUE,SQ_TpmVersion=2,SQ_RamMB=8192,"
> reg.exe add HKLM\SYSTEM\Setup\MoSetup /f /v AllowUpgradesWithUnsupportedTPMOrCPU /t REG_DWORD /d 1
> ```
> 
> I'll add this batch script to the created media in a future version of Rufus.

Seems like you wrongfully included "CompatMarkers" in the second and following paths.

JimBim89, 2024-10-02:
I also needed to put quotation marks around the path of the add command, without that I got wrong syntax.

pbatard, 2024-10-02:
OK, fixed that too. For the record, I tested the above by editing the registry manually, and I am still working on the script that I'm going to have Rufus copy over (which I will test properly of course...)

JimBim89, 2024-10-02:
Thanks, looks good now and is currently installing.

pbatard, 2024-10-02:
> Thanks, looks good now and is currently installing.

Great. Thanks for the help!

100PIER, 2024-10-02:
Thanks pbatard and others members.
Can you confirm me the next RUFUS v4.6 will fix as all the Bypass Windows User Experience options including TPM 2.0 and 'old' CPU (such as Intel I7-5930K on X99 mboard) ? Thanks. 100PIER

pbatard, 2024-10-02:
> Can you confirm me the next RUFUS v4.6 will fix as all the Bypass Windows User Experience options including TPM 2.0 and 'old' CPU

It should fix the new restrictions for in-place upgrade if you **manually** run a batch script that will be placed at the same level as `setup.exe` and that will contain the registry commands posted above.

It will **not** change anything for clean install, since I haven't found breakage in the clean install bypasses while testing, meaning that if a clean install of 24H2 doesn't work with Rufus 4.5, then it also won't work with Rufus 4.6.

For what is worth, I tested in-place upgrade from 23H2 to 24H2 on an Intel I7-5930K/X99 system using the registry trick above, and was able to upgrade without issues. I have not tested a clean install on such a system yet, but I tested on a much older system, and so no issue either.

At any rate, I can **NOT EVER** promise anything with regards to what the next release of Rufus *might* do or not do. You will find out when it is released.

jovan2009, 2024-10-03:
I used the batch file above and I was successful installing on a Ryzen PC with TPM disabled.
But on my old laptop the batch file couldn't help me, still got this screen:
[image]
My CPU is probably really old:
[image]
I thought that's the end of the line. But I went and downloaded Win 11 Boot And Upgrade FiX KiT v5.0, I used option 3 and... it worked.
[image]

Now it's installing 24H2, let's see if my laptop blows up after this :-)
So, long story short, something is missing from batch file to completely bypass requirements, at least in my case.

Juraj-Masiar, 2024-10-03:
@jovan2009 you won't fix this with any hacks, new Windows is compiled with these CPU instructions in use, so it won't boot on these very old CPU.
See for example: https://www.tomshardware.com/software/windows/microsoft-updates-windows-11-24h2-requirements-cpu-must-support-sse42-or-the-os-will-not-boot

jovan2009, 2024-10-03:
@Juraj-Masiar Oh well... It is to be expected. I am on my way to revert to Win 10 anyway, it was the last thing I was going to try with Win 11 before that. Win 11 has become progresively very slow, borderline unbearable, I booted a Win 10 WintoUSB install from a portable SDD a few days ago and it felt like an upgrade a few generations of hardware for my old laptop. So long Win 11, you will not be missed.

100PIER, 2024-10-03:
Hello Pete Batard,

 

Thanks for your help.

I am focused only on in-place W11 23H2 upgrade to W11 24H2 on my ASUS Intel I7-5930K/X99 machine.

As I am not skilled , not familiar and  not aware how to do for « you manually run a batch script that will be placed at the same level as setup.exe and that will contain the registry commands posted above. » , 

Please can you explain me in detail   how you apply that Registry modification  ?

Do you use the RegEdit  Tool ?

Sorry for my ignorance.

At the moment I am stopped on my current W11 23H2 version.

Thanks for your patience.

100PIER

Juraj-Masiar, 2024-10-03:
@100PIER you can also run those commands one by one in an Admin terminal:
1. right click start button and select "Terminal (Admin)" or "PowerShell (Admin)"
2. this will open console window where you can execute them one by one (copy first line, paste, press Enter, then repeat for next line...)

jovan2009, 2024-10-03:
@100PIER 
1. Copy the content of that field above, containing the reg.exe commands
2. Right click on desktop an choose "New -> Text Document"
3. Double click the new text file created
4. Paste what you copied at step 1 into the text file and save it.
5. Rename the text file to whatever you want, changing the extension from ".txt" to ".bat"
6. Right click on the ".bat" file and choose "Run as Administrator" and give all the requested permissions.
7. Run Setup.exe from the Win 11 USB stick made with Rufus

This should work. At least that's what I've done.

100PIER, 2024-10-03:
Many thanks for your help Juraj, 

I do understand the procedure.

I will try soon and let you know the results.

At the moment I have copied the script in a text file format, so I can copy/paste line per line.

Regards,

Alain

pbatard, 2024-10-03:
If you're talking about the last comma, it is very much required as it's a `REG_MULTI_SZ` which must end with an empty entry, and where each value is separated by commas.

If you're talking about the CPU stuff from the Win 11 Boot And Upgrade FiX KiT, that's very much intentional.

stdin82, 2024-10-03:
Those values will not be enough for all machines / CPUs

pbatard, 2024-10-03:
I am not planning to add extra CPU bypasses besides the main bypass from `AllowUpgradesWithUnsupportedTPMOrCPU` (because I am pretty sure that, even if those bypass work, you will run into issues, and I don't want to have to "support" these kind of installations).

wyxchari, 2024-10-03:
I have W11 23H2. I tried to INSTALL W11 24H2 from USB created with Media Creation Tool, then using the trick of creating an empty appraiserres.dll and it works for clean installs.
Now I want to UPGRADE (not clean install) from 23H2 to 24H2 and the appraiserres.dll trick doesn't work. So I overwrote the 23H2 unzipped ISO with the install.wim and boot.wim files from the 24H2 version and then I do the appraiserres.dll trick. Then W11 24H2 installs.

a2line, 2024-10-03:
If it can help, I have applied AIO 2a fix from Enthousiast tool on 24H2 ISO and created an UEFI USB key with Rufus 4.5p. I installed the OS on a empty partition like a charm (the machine is 11 years old, i7 4th gen). As I have checked the five others Rufus options for Windows iso (unchecked the 1st one), it seems that they are all still operational at last, I have the username set, it skipped privacy questions, no Ms account asked…

pbatard, 2024-10-03:
For clean install, all the options of Rufus 4.5 should work. Especially, you should not have to do anything special to clean install 24H2. This thread, and the registry commands, are only about **in-place** upgrades, where you run `setup.exe` from the media on a running Windows installation, rather than boot from USB.

a2line, 2024-10-03:
Thank you, that's good to know, I wasn't sure of that. I can retry with all options, it's very quick. I can also confirm the 5 registry commands are ok to have setup.exe bypass the requirements to upgrade. As that in-place upgrade got stuck at 31 % for two hours on that machine (nearly nothing installed, probably stuck on some old drivers…) I did try a clean install.

wyxchari, 2024-10-03:
It also works by upgrading from 23H2 to 24H2 by putting:
setup /product server
or
\sources\setupprep /product server
There is no need to hack the appraiserres.dll

100PIER, 2024-10-03:
Pete Batard, Juraj, Jovan2009

Many thanks for your help.
Finally I have use the **PowerShell 7** method to modify my current W11 23H2 Registry.
So, for my **Intel I7-5930K/X99** platform I have executed manually line per line the Commands (via copy/paste) extracted of the attached Text File:
[patch_03oct24.txt](https://github.com/user-attachments/files/17249535/patch_03oct24.txt)
Screen copy of the PS commands:
[image]
I have downloaded the W11 24H2   ISO file using the MediaCreationTool (v10.0.26100.1)
Then I have create the USB bootable from RUFUS with these parameters:
[image]
[image]
Here is the RUFUS log:
[rufus_W1124H2x64FR_03oct24.log](https://github.com/user-attachments/files/17249669/rufus_W1124H2x64FR_03oct24.log)
Then I have run (admin) the SETUP.EXE located on the USB created with RUFUS.
The upgrade process does take about 2 hours  !!!, and finally I do run W11 24H2 now and applied the last Windows Update.
I do write this mail under this successful in-place upgrade to W11 24H2.
Thanks again for all help I have received.
100PIER

Gajalshankar, 2024-10-04:
Hy,
I checked using the PC Health check I've only TPM 2.0 and secure boot requirements fulfilled. I'm on i5-7200U(Unsupported) with 16GB RAM. I don't remember how I upgraded from Windows 10 to Windows 11 last time.

[image]
[image]

Is it possible to do an in-place upgrade to Windows 11 24H2 without a clean install?

Gajalshankar, 2024-10-04:
@pbatard Thanks but in my case do I've to create a bootable USB using Rufus with those checks or can I directly mount the ISO after running the bash script ?

pbatard, 2024-10-04:
You have to create an USB with Rufus, then execute the registry commands, then run `setup.exe` from the drive you created.

prashanthkp79, 2024-10-04:
Myself getting the following error after running the registry commands, then run setup.exe from the USB drive. Unable to start with setup

[image]

pbatard, 2024-10-04:
@prashanthkp79, see the comments above. This has already been answered. Your CPU is just too old to be supported by Windows 11 24H2.

prashanthkp79, 2024-10-04:
> @prashanthkp79, see the comments above. This has already been answered. Your CPU is just too old to be supported by Windows 11 24H2.

But i am using Windows 11 23H2 till date for my daily routine. Any help for the upgrade will be much useful.

thecatontheceiling, 2024-10-04:
You use a CPU from 2011 for windows 11? Jeez that must suck...

Gajalshankar, 2024-10-04:
> You have to create a USB with Rufus, then execute the registry commands, then run `setup.exe` from the USB drive you created.

@pbatard 
After creating bootable USB with Rufus with all default options, I executed the following registry commands 
[image]
then I ran `setup.exe`, I got the following pop-up
[image]
[image]
After executing sfc cmd to fix any corrupted files again I ran `setup.exe`.... still that same pop-up 

I thought something might be wrong with USB so I've created USB again... still the same pop up 
I also tried by directly mounting the ISO in that case `setup.exe` worked and as soon as installation started I got this pop-up 
[image]

Unfortunately, nothing worked in my case. Can you help me with this

pbatard, 2024-10-04:
>  Can you help me with this

Most likely you have a security solution that interferes with autorun files (like **poor** security solutions tend to do).

Have you validated that the `source/autorun.dll` you **should** find on your USB matches the one you will find on the ISO.

From where I stand, your issue looks 100% environmental and has absolutely nothing to do with Rufus. Oh and there's no need to start playing with `dism` or `sfc` when you can very easily validate whether the rather explicit error message you get is accurate...

Gajalshankar, 2024-10-04:
@pbatard 
directory H: is USB and I: is mounted ISO
[image]
I got around 61 different DLL files but `autorun.dll` wasn't one of them...  do I've to replace DLLs from ISO to USB?
Can you explain what you mean by poor security solutions?

igorruckert, 2024-10-04:
Still no luck with in-place upgrade, after adding commands when I run setup.exe nothing happens. Investigating the logs I found error 0x80070002. Will attach log if someone could find what is blocking the installation
[BlueBox.log](https://github.com/user-attachments/files/17263581/BlueBox.log)

pbatard, 2024-10-04:
@Gajalshankar 

Man, you really do love the command line... 😄 Was File explorer not available to look in the `sources/` directory of the USB to find out if there was an `autorun.dll` there (and yes, I will take your word for it, no need for screenshots)?

> Can you explain what you mean by poor security solutions?

I mean *ill devised* security solutions that, because there happened to be `autorun.inf` related malware attacks **about 15 years ago** have decided that anything named `autorun` should be considered malware, and actively deleting `autorun.inf` or `autorun.dll` after they have been copied over by legitimate programs.

I can assure you that it's definitely **not** Rufus that is "somehow forgetting" to copy `autorun.dll` here, but something on your computer that is actively preventing it from existing on your USB. And that's the extent of what I can say on the matter. **YOU** will have to figure out why `autorun.dll` is being deleted from your drive.

Gajalshankar, 2024-10-04:
@pbatard It was easier to validate files using the command line 😅 
rufus didn't forget or delete the file from USB. From "autorun.dll wasn't one of them", I meant  `source/autorun.dll` in USB matches with the one I found on the ISO but all other dll weren't matched with the ISO ones. I wonder how my computer is preventing it from existing on USB.. I'll try on a different USB that could be the issue

Thank you for your explaining and helping out.

Legendarion, 2024-10-04:
> You use a CPU from 2011 for windows 11? Jeez that must suck...

Heh, and why should that suck? It's a pro/"pro" to still use it today. Damn man, I have a Phenom CPU myself on my reserve desktop computer and a Phenom II CPU that is still in use today bought in 2012 and that is still rock solid.

pbatard, 2024-10-04:
Please take discussions of what CPUs are worthy to run Windows 11 somewhere else, or I will start deleting posts, as this thread is getting long enough already. These kind of discussions are squarely off-topic.

prashanthkp79, 2024-10-05:
That's right, Phenom still rocks!!!!!

glennmcc, 2024-10-05:
> For in-place upgrades, you need to run the following from an elevated command prompt before running `setup.exe` (credits to [Win 11 Boot And Upgrade FiX KiT](https://forums.mydigitallife.net/threads/win-11-boot-and-upgrade-fix-kit-v5-0-released.83724/)):
> 
> ```batchfile
> reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers" /f 2>NUL
> reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\Shared" /f 2>NUL
> reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\TargetVersionUpgradeExperienceIndicators" /f 2>NUL
> reg.exe add "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\HwReqChk" /f /v HwReqChkVars /t REG_MULTI_SZ /s , /d "SQ_SecureBootCapable=TRUE,SQ_SecureBootEnabled=TRUE,SQ_TpmVersion=2,SQ_RamMB=8192,"
> reg.exe add "HKLM\SYSTEM\Setup\MoSetup" /f /v AllowUpgradesWithUnsupportedTPMOrCPU /t REG_DWORD /d 1
> ```
> 
> I'll add this batch script to the created media in a future version of Rufus.

Thank you very much, it worked perfectly and the in-place upgrade from 23h2 to 24h2 is currently underway.

pbatard, 2024-10-06:
Note that, with the next release of Rufus, you'll simply be able to double click on `setup.exe` to perform in-place upgrades, as Rufus will create a `setup.exe` wrapper that applies the registry bypasses before running the original Microsoft setup.

If you are in a hurry to test this, you can just simply extract an ALPHA release of the upcoming Rufus from the MinGW artifacts available at https://github.com/pbatard/rufus/actions/runs/11201946687.

cowbutt, 2024-10-07:
> It also works by upgrading from 23H2 to 24H2 by putting: `setup /product server` or `\sources\setupprep /product server` There is no need to hack the appraiserres.dll

I can confirm this worked to upgrade my i7 5820K system from 23H2 to 24H2 today, using an ISO created with the Media Creation Tool.

The only tiny issue I noticed was that Gigabyte's (?) gdrv.sys driver is blocked from loading, I believe because the signing certificate has been revoked because the driver is vulnerable to exploitation.

thecatontheceiling, 2024-10-07:
I just read the new setup shim and it's cancer

Why are you doing the malware trick of running CreateProcess on a .dll?

If you're making a shim anyway you can just shim HwReqChk.dll instead, it's much more simpler

https://github.com/asdcorp/hwreqchk

100PIER, 2024-10-07:
Thanks cowbutt,

On my side, I used the « PowerShell » line/line method to modify the W11H2 Registry.

Your method does seem quite simpler, however I did not used it because I do ignore if it was compatible to upgrade a Windows Desktop Family version.

On my ASUS Sabertooth X99 desktop (i7-5930K), the only issue I noticed was that ASUS driver « asio.sys » is blocked from loading under W11 24H2 (Family Version) by the « Assistant Compatibility Program ». 

It was not blocked under W11 23H2.

For the moment I do ignore this non loading driver warning message..

pbatard, 2024-10-07:
@thecatontheceiling

> Why are you doing the malware trick of running CreateProcess on a .dll?

From https://github.com/pbatard/rufus/tree/master/res/setup#readme:

> Obviously, the fact that we "inject" a setup executable may leave people uncomfortable about the possibility that we might use this as a malware vector (...)

I'm using **tons** of malware-like tricks in Rufus (inluding ones inspired by actual BlackLotus malware to download the diskcopy.dll from the Windows debug servers), so if you wanna start complaining about this one, you haven't looked that closely at what Rufus does.

>  the process for opening the setup is now: setup.exe -> setup.dll -> setupprep.exe -> setuphost.exe

From setup.exe -> setupprep.exe -> setuphost.exe. Big deal.

> you can just shim HwReqChk.dll instead

Except I fully expect Microsoft to alter the way they check for requirements in the future, which is why I'm taking provision to handle setup at the source rather than in a DLL that may or may not get removed/renamed/altered. That way, if things change downlevel, I'm not going to have to redo a whole new sub-project and workflow and should be able to react to changes swiftly.

> You can't just delete setup.exe completely either as that will cause issues unless you reimplement the entire thing

Please elaborate in that. Why would one want to delete `setup.exe`? I don't understand what you're trying to get at here.

pbatard, 2024-10-07:
Also, I need to point out that, where possible, I want to use Microsoft implemented bypasses, rather than hack DLLs, so what I really want here is apply registry entries that were made available by Microsoft. So the solution I really wanted here is one where the required registry keys are created/deleted so that setup will proceed (but still filter out CPUs that lack PopCnt, SSE4.2, etc.) and not a _"let everything through and let users sort it out"_. Which means that I had 3 ways to go about that::

* Ask users to manually run a batch file with elevation to apply the registry changes (annoying for users who may have trouble with elevation) before running `setup.exe`.
* Ask users to manually run an exe that performs elevation and applies the registry changes, before running `setup.exe`.
* Just replace `setup.exe` and make the whole thing as user-friendly as possible.

Of course, replacing the `HwReqChk.dll` would also have accomplished the same goal as option 3, but it would have failed the goal of keeping with the bypass range that Microsoft set up (unless we are to deduplicate part of what the original `HwReqChk.dll` did, which is way more than I want to invest here).

DaneshCode, 2024-10-07:
> Also, I need to point out that, where possible, I want to use Microsoft implemented bypasses, rather than hack DLLs, so what I really want here is apply registry entries that were made available by Microsoft. Which means the solution I aim for is one where the required registry keys are created/deleted so that setup will proceed (but still filter out CPUs that lack PopCnt, SSE4.2, etc.) and not a _"let everything through and let users sort it out"_. Which means that I had 3 ways to go about that::
> 
> * Ask users to manually run a batch file with elevation to apply the registry changes (annoying for users who may have trouble with elevation) before running `setup.exe`.
> * Ask users to manually run an exe that performs elevation and applies the registry changes, before running `setup.exe`.
> * Just replace `setup.exe` and make the whole thing as user-friendly as possible.
> 
> Of course, replacing the `HwReqChk.dll` would also have accomplished the same goal as option 3, but it would have failed the goal of keeping with the bypass range that Microsoft set up (unless we are to deduplicate part of what the original `HwReqChk.dll` did, which is way more than I want to invest here).

**Does this mean there's no way for systems that lack SSE4.2 and PopCnt to install the 24H2 update?**

DaneshCode, 2024-10-07:
>  (but still filter out CPUs that lack PopCnt, SSE4.2, etc.) 
> 

Does this mean there's no way for systems that lack SSE4.2 and PopCnt to install the 24H2 update?

pbatard, 2024-10-07:
@DaneshCode, sorry for deleting one your double posted comment when you deleted the other one! You got your answer above.

xosevp, 2024-10-07:
> > **Does this mean there's no way for systems that lack SSE4.2 and PopCnt to install the 24H2 update?**
> 
> From what ive heard, attempting to install 24H2 on a system without SSE4.2/PopCnt will result in BSOD

A [x86-64-v2](https://en.wikipedia.org/wiki/X86-64#Microarchitecture_levels) CPU supporting [SSE4.2](https://en.wikipedia.org/wiki/SSE4.2) and [POPCNT](https://en.wikipedia.org/wiki/SSE4#POPCNT_and_LZCNT) CPU instructions is now required: Intel Nehalem(~2009) , AMD Bulldozer(~2012), or better.
Otherwise the Windows kernel is unbootable. See: https://twitter.com/abc/status/1781922187252797805

Gajalshankar, 2024-10-08:
@pbatard 
Hi, I've changed the USB... now there are no errors related to `autorun.dll` or anything while running `setup.exe` and when I reach the last page (with keeping files and apps) as soon as the installation starts, it says -
[image]
[rufus.log](https://github.com/user-attachments/files/17294328/rufus.log)

 I've also tried using the alpha release..., but that also didn't work :(

pbatard, 2024-10-08:
> when I reach the last page (...) as soon as the installation starts, it says - `Windows 11 installation has failed`

Then it just means that your system is not compatible with an upgrade. Neither Rufus nor Microsoft can do magic with regards to the requirements of the installer or the new requirements of 24H2. You may find some information as to why your upgrade failed in the [Windows setup logs](https://support.microsoft.com/en-gb/topic/log-files-that-are-created-when-you-upgrade-to-a-new-version-of-windows-9ec8aa31-0cc1-a0b2-2d98-e9c6714349b9) but please don't expect this issue tracker to supersede Microsoft when it comes to your specific problem. We don't have access to the setup/upgrade source code. We don't know what's happening behind the scenes that makes your installation fail.

extrimexxx, 2024-10-08:
> If you are in a hurry to test this, you can just simply extract an ALPHA release of the upcoming Rufus from the MinGW artifacts available at https://github.com/pbatard/rufus/actions/runs/11201946687.

@pbatard Hi! if enable runtime uefi media validations

[image]

md5sum of setup.exe in md5sum.txt going from orig exe, not from modified

[image]

pbatard, 2024-10-08:
@extrimexxx, thanks for the report. I indeed forgot to add the new `setup.exe` to the list of files that need to be rehashed for `md5sum.txt`.

extrimexxx, 2024-10-08:
> @extrimexxx, thanks for the report. I indeed forgot to add the new `setup.exe` to the list of files that need to be rehashed for `md5sum.txt`.

md5sum_totalbytes also needed "recalculate"  :)

Daopa-2, 2024-10-08:
P Batard, hello.  I am typing at you on a no upgradable Dell. Thanks to you, it is now on W-11. I envy guys like you who know so much in the innards of our machines. I wish I were 20 again, ( I am 92 ) and pick up and remember all this stuff. I pulled a fast on doing the upgrade,  I cut & pasted the 5 command lines to a .doc and got the file opened on a split screen, right half. Then on the left half, I got the CMD screen and again just C&P every of the 5 CDM lines from one to the other.  Worked like a charm. I was just trying and it worked. Thanks for the guidance, it is appreciated.
There were a few moments when I had doubts, it was really time taking but fear not, I have lots of time. 
Have a great day.  Daopa-2

PhilipYip1988, 2024-10-09:
Just in case it is useful, Microsoft have also posted an article:

[Microsoft: Ways to Install Windows 11](https://support.microsoft.com/en-gb/windows/ways-to-install-windows-11-e0edbbfb-cfc5-4011-868b-2ce77ac7c70e)

> Microsoft recommends against installing Windows 11 on a device that does not meet the Windows 11 minimum system requirements. If you choose to install Windows 11 on a device that does not meet these requirements, and you acknowledge and understand the risks, you can create the following registry key values and bypass the check for TPM 2.0 (at least TPM 1.2 is required) and the CPU family and model.

> Registry Key: HKEY_LOCAL_MACHINE\SYSTEM\Setup\MoSetup

> Name: AllowUpgradesWithUnsupportedTPMOrCPU

> Type: REG_DWORD

> Value: 1

I tested this and performed an upgrade install by mounting the ISO and running the setup.exe. I've not tested creating a Bootable USB with Rufus and running the setup from it with the above registry edit yet. I'm not too sure what the difference between this registry edit is and the one posted above but it may be useful to include this one as Microsoft have documented it.

Gajalshankar, 2024-10-09:
@pbatard 
C:\Windows\Logs\MoSetup\BlueBox.log
[Bluebox.log](https://github.com/user-attachments/files/17304824/Bluebox.log)
C:\$Windows.~BT\Sources\panther\setupact.log
[setupact.log](https://github.com/user-attachments/files/17304825/setupact.log)

[36 more replies]

closed 2024-10-10 by pbatard