pbatard/rufus #2573: Changes to GPT Header

chazste, 2024-10-07:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Writing compress GPT image results in unexplained change to the "Starting LBA of array of partition entries (usually 2 for compatibility)". Offset 0x48 in EFI partition header.

rufus started from command line with rufus -g -i foo.bz2

- foo.bz2 created on Linux using losetup/fdisk/mkfs/tar
- foo.bz2 compressed using pbzip2
- original image size is 17 GiB
- In addition to the change to the header the GPT gets written (again) into the "slack" space between the LBA 1 and the start of the first partition. This overwrites part of u-boot bootloader which is stored in the slack space.
- Checking the resulting SD card on Linux using fdisk shows the following messages/warnings

```
## result of rufus image to 32 GB card. 
# fdisk /dev/sda

Welcome to fdisk (util-linux 2.37.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

GPT PMBR size mismatch (35651583 != 62521343) will be corrected by write.

Command (m for help):
```
 
- writing the same image from Linux using a combination of pbzip2 and dd works, but yields different warnings when running fdisk on Linux.
```
## result of dd of image to a 32GB card. No fixup of PMBR and backup GPT in wrong place.

# fdisk /dev/sda

Welcome to fdisk (util-linux 2.37.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

GPT PMBR size mismatch (35651583 != 61067263) will be corrected by write.
The backup GPT table is not on the end of the device. This problem will be corrected by write.

Command (m for help):
```

Log
---
```
Rufus x64 v4.4.2103 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19045.4894)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: OpenLocalMachineGPO failed - error 80004005
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found USB 2.0 device 'Generic Flash HS-CF USB Device' (0424:2228)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Found USB 2.0 device 'Generic Flash HS-COMBO USB Device' (0424:2228)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitione
![screenshot](https://github.com/user-attachments/assets/f675233d-40cb-4349-adae-404c26e2c467)
d

Image provided: 'E:\projects\psgg\images\foo.bz2'
Scanning image...
ISO analysis:
  'E:\projects\psgg\images\foo.bz2' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: foo.bz2 (3 GB)
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Will use 'Z:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Requesting logical volume handle...
Writing compressed image:
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive2 for shared write access

Found USB 2.0 device 'Generic Flash HS-CF USB Device' (0424:2228)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Found USB 2.0 device 'Generic Flash HS-COMBO USB Device' (0424:2228)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 5
Disk GUID: {3FEF1265-D906-464A-9795-1FEED1364BA6}
Max parts: 128, Start Offset: 1048576, Usable = 32009862656 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot'
  Detected File System: FAT16
  ID: {00A48EE8-A587-A741-8ED0-B5E06532344F}
  Size: 200 MB (209715200 bytes)
  Start Sector: 4096, Attributes: 0x0000000000000000
Partition 2:
  Type: Linux Data Partition
  Name: 'root'
  Detected File System: ext4
  ID: {0EA21BED-09A7-6F4C-8554-36B5B5DAE762}
  Size: 6 GB (6442450944 bytes)
  Start Sector: 413696, Attributes: 0x0000000000000000
Partition 3:
  Type: Linux Data Partition
  Name: 'audit'
  Detected File System: ext4
  ID: {C618EF46-70A8-0D4F-B925-7FE7782AD22A}
  Size: 250 MB (262144000 bytes)
  Start Sector: 12996608, Attributes: 0x0000000000000000
Partition 4:
  Type: Linux Data Partition
  Name: 'data'
  Detected File System: ext4
  ID: {19A3870E-537A-D544-945C-D5F01F0AB5CA}
  Size: 1.6 GB (1673510400 bytes)
  Start Sector: 13508608, Attributes: 0x0000000000000000
Partition 5:
  Type: Linux Data Partition
  Name: 'xfer'
  Detected File System: ext4
  ID: {6624A410-117A-E948-AA31-4348B2E5F01B}
  Size: 9.0 GB (9663659520 bytes)
  Start Sector: 16777216, Attributes: 0x0000000000000000
Found USB 2.0 device 'Generic Flash HS-CF USB Device' (0424:2228)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Found USB 2.0 device 'Generic Flash HS-COMBO USB Device' (0424:2228)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 5
Disk GUID: {3FEF1265-D906-464A-9795-1FEED1364BA6}
Max parts: 128, Start Offset: 1048576, Usable = 32009862656 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot'
  Detected File System: FAT16
  ID: {00A48EE8-A587-A741-8ED0-B5E06532344F}
  Size: 200 MB (209715200 bytes)
  Start Sector: 4096, Attributes: 0x0000000000000000
Partition 2:
  Type: Linux Data Partition
  Name: 'root'
  Detected File System: ext4
  ID: {0EA21BED-09A7-6F4C-8554-36B5B5DAE762}
  Size: 6 GB (6442450944 bytes)
  Start Sector: 413696, Attributes: 0x0000000000000000
Partition 3:
  Type: Linux Data Partition
  Name: 'audit'
  Detected File System: ext4
  ID: {C618EF46-70A8-0D4F-B925-7FE7782AD22A}
  Size: 250 MB (262144000 bytes)
  Start Sector: 12996608, Attributes: 0x0000000000000000
Partition 4:
  Type: Linux Data Partition
  Name: 'data'
  Detected File System: ext4
  ID: {19A3870E-537A-D544-945C-D5F01F0AB5CA}
  Size: 1.6 GB (1673510400 bytes)
  Start Sector: 13508608, Attributes: 0x0000000000000000
Partition 5:
  Type: Linux Data Partition
  Name: 'xfer'
  Detected File System: ext4
  ID: {6624A410-117A-E948-AA31-4348B2E5F01B}
  Size: 9.0 GB (9663659520 bytes)
  Start Sector: 16777216, Attributes: 0x0000000000000000

```

chazste, 2024-10-07:
[image]

pbatard, 2024-10-07:
I'll look into it, but one thing you should know is that any flat image (compressed or not) that uses GPT as its partition scheme will result in an invalid GPT, because GPT requires a secondary GPT to resides at `<end of disk> - 23 sectors`, and of course, unless you are using an image that was designed for the exact size of the disk you are writing it to (which, for 99% os usage is never the case), then, once Rufus or dd or balenaEtcher are done writing your image, you will have garbage in the secondary GPT which needs to be fixed.

I believe this is part of the error you see with `GPT PMBR size mismatch (35651583 != 62521343) will be corrected by write.`

So the first thing I would advise is to test what happens when using plain `dd`, especially as the dd writing code of Rufus is fairly straightfoward (as can be expected) and last time I ran checks to validate that what Rufus writes is exactly what is provided as source, I validated that there was not a single byte that was different (that is, unless you have a partition that Windows can mount, because then it may create a `System Volume Information\` subdirectory, with a bunch of files, which of course will alter the content, but that's Windows behaviour, nor Rufus).

chazste, 2024-10-07:
@pbatard 
This is a little weirder than I was expecting. 
- Using dd for windows; even though the write does not complete for all 18 GB. The field at offset 0x48 is not 2
- Using balenaEtcher I managed to snatch the card out of my PC when offset 0x48 was 2 (correct) then when I put it back in the Windows PC and checked the partition table with diskpart it became wrong.

So my conclusion is this is not a bug in rufus. It is some bad interaction between my image and Windows idea of a good GPT.

So honestly I think you can close this issue.

pbatard, 2024-10-09:
Ok, I'm going to close this issue then.

closed 2024-10-09 by pbatard

chazste, 2024-10-17:
As a hint to my future self or anyone else who may happen along wondering why Windows silently changes the location of the primary GPT partition array...

It seems the key is offset 0x28. Which Wikipedia describe as

40 (0x28) | 8 bytes | First usable LBA for partitions (primary partition table last LBA + 1)
-- | -- | --

Microsoft seems to take the parenthetical note as an imperative and moves the partition array to match, while updating the field to indicate the beginning of the array. When creating a GPT disk Microsoft sets this field to 0x22 or the first sector after a 128 entry partition array. Without additional guidance Windows will create the first partition at 1 MiB. Windows requires that partitions be aligned on 1 MiB boundaries. 

fdisk from util-linux 2.37.2 seems to prefer the statement portion and sets this filed to 0x0800 (2048) or 1 MiB into the disk