pbatard/rufus #2574: Can't remove TPM 2.0 requirement from Windows 11 24H2 using Rufus

emulationdude, 2024-10-07:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I'm trying to create a Windows 11 24H2 installer using Rufus. I checked the box to remove RAM/TPM 2.0 requirement. However, the resulting installer still requires a TPM 2.0 chip and will not advance on a system without it. I have used Rufus in the past to bypass this requirement as of Windows 11 23H2. Is it no longer possible to remove that requirement?>

Log
---
Log is attached
[rufus.log](https://github.com/user-attachments/files/17283271/rufus.log)

```

pbatard, 2024-10-07:
[Answered in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_setup_says_my_PC_doesnt_meet_Windows_11_system_requirements).

closed 2024-10-07 by pbatard

JiiPee74, 2024-10-10:
> [Answered in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_setup_says_my_PC_doesnt_meet_Windows_11_system_requirements).

However there is small error in wiki and delimiter is missing quotes so it does not work.

Also '2>NUL' seems to give error in powershell at least.

So I would suggest this:
```
reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\CompatMarkers" /f
reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\Shared" /f
reg.exe delete "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\TargetVersionUpgradeExperienceIndicators" /f
reg.exe add "HKLM\SOFTWARE\Microsoft\Windows NT\CurrentVersion\AppCompatFlags\HwReqChk" /f /v HwReqChkVars /t REG_MULTI_SZ /s "," /d "SQ_SecureBootCapable=TRUE,SQ_SecureBootEnabled=TRUE,SQ_TpmVersion=2,SQ_RamMB=8192,"
reg.exe add "HKLM\SYSTEM\Setup\MoSetup" /f /v AllowUpgradesWithUnsupportedTPMOrCPU /t REG_DWORD /d 1
```

pbatard, 2024-10-10:
The script is not designed to be run from PowerShell. And at any rate, Rufus 4.6 (of which you can find a BETA [here](https://rufus.ie) will apply the registry changes for you, so you don't have to run the script.