pbatard/rufus #2575: Write error at sector 62324736: [0x0000001B] The drive cannot find the sector requested.

santosdanillo, 2024-10-08:
﻿

Checklist
---------
- [ x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When writing a .vhdx image to a SDCard I see the following error:
Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Write error at sector 62324736: [0x0000001B] The drive cannot find the sector requested.
Retrying in 5 seconds...

Write error at sector 62324736: [0x0000001B] The drive cannot find the sector requested.
Retrying in 5 seconds...

Write error at sector 62324736: [0x0000001B] The drive cannot find the sector requested.
Retrying in 5 seconds...

Write error at sector 62324736: [0x0000001B] The drive cannot find the sector requested.

The SDCard that was cloned is the same model (SanDisk Ultra) and size (32Gb) as the 2 fresh SDCards I tried flashing the .vhdx to.

SAVING SDCARD TO .VHDX IMAGE LOG:

Rufus x64 v4.5.2180 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.4894)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x69043244
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 3.9 GB (4227826176 bytes)
  Start Sector: 63, Boot: No
Partition 2:
  Type: QNX (0xb1)
  Detected File System: (Unrecognized)
  Size: 25.8 GB (27683389440 bytes)
  Start Sector: 8257536, Boot: Yes

[... 229 lines cut ...]
Re-mounted volume as D: after error

Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x69043244
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 3.9 GB (4227826176 bytes)
  Start Sector: 63, Boot: No
Partition 2:
  Type: QNX (0xb1)
  Detected File System: (Unrecognized)
  Size: 25.8 GB (27683389440 bytes)
  Start Sector: 8257536, Boot: Yes
Saved rufus-write-vhdx-to-sdcard-FAIL-0x0000001B.log

Computing hash for 'A5-sdcard.vhdx'...
  MD5:    551575b9fd4d55deb15d1be4ad8edb25
  SHA1:   6f26d3dd56d9305a8bef443eece21a6b42b30599
  SHA256: 5f6ae4b23aa21fbc37916ca264d2e2f415d6d1568177193f699cfdd43f432003

pbatard, 2024-10-08:
> - [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.

Please provide a log that shows that you _actually_ did that.

pbatard, 2024-10-10:
I'm going to assume that OP didn't run the bad blocks check and that, if they did, they would have found that their drive was defective, and therefore close this issue.

closed 2024-10-10 by pbatard

santosdanillo, 2024-10-10:
A bit more time, the new sdcards are in the office and I am WFH last couple
of days.