pbatard/rufus #2579: windows 10 mount problem

haddubrandt, 2024-10-11:
when i mount the latest windows 10 iso in rufus, it requires online microsoft account.
please add that windows user experince popup.

pbatard, 2024-10-11:
You just need to unconnect the network. And you have the option to create a local account.

The option for Windows 11 is only proposed because, unlike Windows 10, you can't simply disconnect the network and have Windows setup allow you to use a local account.

In other words, there is no need to add any new option for Windows 10.

closed 2024-10-11 by pbatard

haddubrandt, 2024-10-11:
i already offline. but latest windows 10 22h2 aug/sept iso's have not continue without internet connection.
if you can test, you will see..

haddubrandt, 2024-10-11:
i download the W10 consumer version from massgrave or rg-adguard.
internet connection requirement issue has started after the August ISO update.
even though I was installing offline, they made this mandatory with the latest update.
i wanted you to be able to add this feature to the user experience popup you made in windows 11.

pbatard, 2024-10-11:
Yeah, it depends. If I find time to test this, and it's easy to test the Windows 10 version that requires it, I might add the extra options. But, indeed, with Windows 10 reaching EOL soon, don't get your hopes up.

haddubrandt, 2024-10-12:
> Yeah, it depends. If I find time to test this, and it's easy to test the Windows 10 version that requires it, I might add the extra options. But, indeed, with Windows 10 reaching EOL soon, don't get your hopes up.

you can test this ISO. this is latest official release.

en-us_windows_10_consumer_editions_version_22h2_updated_sep_2024_x64_dvd_377ddb19.iso

pbatard, 2024-10-12:
> you can test this ISO. this is latest official release.

Tested with `en-gb_windows_10_consumer_editions_version_22h2_updated_sep_2024_x64_dvd_377ddb19.iso`, with network disconnected, and saw no issue creating a local account during setup.

I therefore have to conclude that you didn't test properly, because my testing shows that updated Windows 10 22H2 images still do **not** require a Microsoft online account and behave just as they have behaved in the past.