pbatard/rufus #2581: RUFUS4.6  will be detected as a virus.

Shima17320508, 2024-10-11:
﻿

pbatard, 2024-10-11:
* https://www.virustotal.com/gui/file/a347261ba24c0076acc91de0aa29486695c7620a6b00230acfc2781eb8413a6f/detection
* https://github.com/pbatard/rufus/wiki/FAQ#user-content-Antivirus_X_reports_that_Rufus_contains_malware
* https://github.com/pbatard/rufus/wiki/Security#general

For Rufus to be detected as a virus, it would have to actually run malware, which it doesn't. And since it doesn't do that, **any** report of Rufus being detected as a virus will be easy to resolve as false positive by contacting the security solution vendor.

closed 2024-10-11 by pbatard

pbatard, 2024-10-11:
Oh, and since we're playing, I'm just going to leave this here, in case I need to reference it in the future...
```
765bc6fb0c2c8e5d4de0e6b94a1e595f
```