pbatard/rufus #2583: The 4.6 Beta triggered Avast and placed several svcHOST files in my USER/public directory

Michaelmyers1958, 2024-10-13:
﻿After downloading the Beta I attempted to make an image. My antivirus went off and added the files in the title. I can't remember the exact wording of the error but it said something about a website. I am sorry but I have no logs because I had to reinstall my OS to get rid of that continuing error message. 

Checklist
---------
- [x ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ x] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

Michaelmyers1958, 2024-10-13:
After downloading the Beta I attempted to make an image. My antivirus went off and added the files in the title. I can't remember the exact wording of the error but it said something about a website. I am sorry but I have no logs because I had to reinstall my OS to get rid of that continuing error message.

pbatard, 2024-10-13:
Also, since there are multiple malicious clones of our website, that attempt to peddle malware, please be sure to only download Rufus from https://rufus.ie or this repository.

If you downloaded it from a different website, you probably downloaded malware.

closed 2024-10-13 by pbatard