pbatard/rufus #2584: Disabling Bitlocker doesn't work

abecedad, 2024-10-14:
﻿

Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi, I'm talking about Rufus v4.6 Beta, when creating the Windows 11 24H2 (final public release) installer, even if I check the "Disable Bitlocker" option, it doesn't matter because the system is still encrypted as soon as the installation is finished.

Thanks.

pbatard, 2024-10-14:
I **still** want to see the log of how you created the media.

I'm sorry to be rude, but is there a part of the following that I should rewrite so that people understand that they MUST produce a log, even for the kind of issues that you opened?:

> - [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.

pbatard, 2024-10-14:
Well, that's how Microsoft officially documents it in: https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-bitlocker#disable-bitlocker-automatic-device-encryption. So that would mean that Microsoft's official documentation has become obsolete.

As long as someone doesn't do what happened in #2121 (which is where the whole feature originated) and can point to what should be used to disable Bitlocker for 24H2 _if_ it doesn't currently work (which I don't recall happening to me when I tested a clean install of 24H2, but I haven't really looked that closely), the current feature will pretty much remain as is, since it is still officially how Microsoft indicates BitLocker should be disabled...

pbatard, 2024-10-15:
> All of the devices i have observed this behaviour on were ones that shipped with Windows 11 Pro from the factory (and they all support device encryption with 23H2)

Hmmm, I gotta wonder if maybe Microsoft doesn't leave a permanent UEFI variable behind, just as they do for S-Mode, that forces device encryption then, because:

> I have only tested with using a local account

comes in contradiction to what Microsoft's doc says should happen: _"(Automatic device encryption) is enabled only after users sign in with a Microsoft Account"_.

Unfortunately, since I don't have a TPM enabled machine I can sacrifice for this, and I assume that Microsoft's documentation of TPM being required for DE is _still_ accurate, I cannot test on a machine that never had encryption applied to find out if there's some truth to a device having previously used DE being automatically enrolled for DE (or, a device not having previously used DE **not** being automatically enrolled for DE). And yeah, it's surprising that Microsoft would encrypt a disk when not using a Microsoft account, because then one has to wonder where the heck, if the user was not prompted to save it, one can get a recovery key...

pbatard, 2024-10-22:
I am going to defer this issue for now, until I am in a position to actually test it.

closed 2024-10-30 by pbatard

JonnyTech, 2024-10-30:
FYI, this is a thing: [Automatic BitLocker Device Encryption for Dell Computers](https://www.dell.com/support/kbdoc/en-us/000124701/automatic-windows-device-encryption-bitlocker-on-dell-systems). Very annoying when I install Windows. Some of my older devices have an option to disable it in BIOS (or UEFI or boot or system or whatever it is called nowadays) setup. And it is not just Dells. Surface Pro's too IIRC.

pbatard, 2024-10-30:
Yes, that's what I see with 24H2 **when not using the Disable BitLocker Encryption in Rufus**:

> After a clean installation of Windows 11 or Windows 10 is completed and the out-of-box experience (OOBE) is finished, the computer is prepared for first use. As part of this preparation, BitLocker device encryption is initialized on the Operating System drive and fixed data drives.

And I do not see it when the setting is enabled, leading me to think that it does work as advertised.