pbatard/rufus #2587: Rufus 4.5+ not making a proper uefi:ntfs partition

sunkan53, 2024-10-15:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When I try to create a bootable usb on Rufus 4.5 and higher, bootable usb is not available as a boot option in uefi. Seems like it formats UEFI:NTFS as RAW filesystem. Rufus 4.4 formats it in FAT12, so it is recognizable in boot menu. 

[rufus.log](https://github.com/user-attachments/files/17384577/rufus.log)
As you can see in the log, used version is 4.6 beta and right before it made a bootable usb in version 4.4. After running it on 4.6 and 4.5 included, it gets formatted to RAW.

Log
---
Character limit was exceeded, so I attached a file
[rufus.log](https://github.com/user-attachments/files/17384592/rufus.log)

pbatard, 2024-10-15:
Your issue is clearly environmental, as I'm not seeing this at all, testing in the same conditions as you, with the same ISO, and on a similar sized drive.

Also, if this was a widespread issue, I'd have had scores of issues similar to yours in this issue tracker, since the release of Rufus 4.5 in May, as Rufus is downloaded close to 3 million times each month (and a lot of these will be for the creation of Windows installation drive that do require UEFI:NTFS, so if Rufus systematically failed to create the UEFI:NTFS partition, it would be obvious by now).

Here is the relevant excerpt from my log:
```
Rufus x64 v4.6.2205 (Beta)
Windows version: Windows 11 Pro for Workstations x64 (Build 26100.2033)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB device 'SanDisk Extreme USB Device' (0781:5580) [ID]
Using 'autorun.inf' label for drive F: 'Ubuntu 24.04 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {77181C81-DC23-4CEE-AF7E-6D59B8862062}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {425361C5-B29E-475D-91C4-4B25E5AEEE10}
  Size: 14.9 GB (16011780096 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {BAA37449-DE71-493D-8730-E739FD2EFEA1}
  Size: 1 MB (1064448 bytes)
  Start Sector: 31275090, Attributes: 0x9000000000000000
Found 5 additional UEFI revocation filters from remote SBAT
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 26100.1742)
  Size: 5.4 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_11_consumer_editions_version_24h2_x64_dvd_1d5fcad3.iso (5.4 GB)
Selected Windows User Experience options:
• Bypass SB/TPM/RAM
• Bypass online account requirement
• Disable data collection

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive7 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
● Creating UEFI:NTFS Partition (offset: 16012846080, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
[... 183 lines cut ...]
  Detected File System: NTFS
  ID: {22DF4333-C50C-4638-A947-20F1B5BCE96F}
  Size: 14.8 GB (15936204800 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {A248DC05-AF3C-4D9F-944F-D2274D79A307}
  Size: 1 MB (1064448 bytes)
  Start Sector: 31127481, Attributes: 0x9000000000000000
```
i.e. with a UEFI:NTFS partition with a file system that was properly recognized.

__Therefore__ your issue has to be environmental.

If I were to take a guess, I'd say you have a security solution on your system, that is interfering with the creation of the UEFI:NTFS partition, and that used to whitelist Rufus 4.4. Weird _"Basic feature F used to work with Rufus x.y but not with newer versions"_ that only **YOU** seem to be having, usually boil down to security solution whitelisting, and a failure to update the whitelist for newer versions of Rufus.

You may also want to take a look at what actually gets written at offset `15937270272` (or `0x3B5EF7200` in hex) on your media (you can use a utility like [HxD](https://mh-nexus.de/en/hxd/) to do just that) and see if it matches the binary content we have in https://github.com/pbatard/rufus/blob/master/res/uefi/uefi-ntfs.img, since all Rufus does is copy every byte from `uefi-ntfs.img` to that location, unmodified.

Especially, you should see something like this:
[image]
With an explicit `UEFI_NTFS` and `FAT12` set of strings at the right location.

If you don't see that, then, since Rufus did not report an error during the write process, it means that **something external to Rufus** interfered with the creation of the UEFI:NTFS partition, and you will have to figure out what it is.

pbatard, 2024-10-20:
No update &rarr; Closing

closed 2024-10-20 by pbatard

sunkan53, 2024-11-04:
Sorry for being late, but I have checked my drive with HxD and offset 

> 0x3B5EF7200

does not seem to be there. 

[image]
 
As you can see, final offset is 0x3B5DF2FF0. By some reason that "UEFI:NTFS" partition is not being created or is inaccessible.
I have added Rufus and its process in Windows Security whitelist (since it is the only antivirus on the device) and I do not have any listed as incompatible software installed.

Guess I will continue to use v4.4. And there is a question: is it okay to use v4.4 for Windows 24H2 and further ISO releases? Could some problems occur using v4.4?

pbatard, 2024-11-04:
You need to open the **Physical disk** in HxD, and **NOT** the Logical disk.

Please look at the title of your window, and compare it to mine. It needs to say "Hard disk <SOME NUMBER>". Else you opened the main partition, which of course does not include the UEFI:NTFS data.

sunkan53, 2024-11-04:
Okay, my bad. I have switched to physical drive view and here is what i got:

[image]

Looks like "UEFI:NTFS" is being written at the wrong offset. Maybe I should try to full format my usb drive?

baigmirza09, 2024-11-29:
# [Download Here Rufus 4.5 Full Free 2025 🔽💹⏬](https://corlubar.com/dl/)