pbatard/rufus #2588: On the same machine, the Rufus' "Windows User Experience" window only appears during the first installation of Windows 11 since the medium was formatted

Goulain-Malbrut, 2024-10-16:
- [x ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Issue description
-----------------
Hello.  On the same machine, the Rufus' _Windows User Experience_ window only appears during the **very first installation of Windows 11** v. 24H2 that is performed since Rufus v. 4.5 formatted the installation medium. Therefore, during subsequent installations on the same machine, it no longer appears; just after the Windows 11 logo is displayed, the _Windows User Experience_ window seems to be inhibited, then the installation process of Windows begins.

pbatard, 2024-10-16:
1. You don't get to pick the parts you want from the check list and the part's you don't.
2. The Wue dialog is from Rufus, not from the Windows installer. Your query makes no sense.
3. You had been warned about not providing a log.

closed 2024-10-16 by pbatard

Goulain-Malbrut, 2024-10-16:
"_You don't get to pick the parts (....)_".

As you can see for yourself, it is nevertheless made possible in your report presentation model.

"_The Wue dialog is from Rufus, not from the Windows installer._"

So what? It is clearly stated in my message. If it pleases you to repeat in your own words what others write; so be it, to each his own style.

"_You had been warned about not providing a log._"

For sure, it cannot be missed in this same model; it is so inadequate when it is off topic. Unintentionally amusing however.

What we can learn is that you must not understand everything you read, this being undoubtedly indicated by your "_Your query makes no sense_.". It was however as simple to interpret as that; my expectation is that _each iteration of installation must be preceded by the exhibition of this Windows User Experience_ window.

And now, what would it do to you to demand a log? That for sure, it would make no sense. So you did well to close, it is still the best thing you could do given your capacity for interpretation. And as if that wasn't eloquent enough, look, there are even some who see in your reasoning a reason to give you a _like_. Ah, it's better to laugh about it, isn't it?