pbatard/rufus #2590: zstd support

guoh27, 2024-10-18:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

This is a feature request. Can we support **zstd** compressed images, similar to bz2?

pbatard, 2024-10-19:
For this to happen, and since decompression in Rufus depends on [Bled](https://github.com/pbatard/bled) which itself depends on [BusyBox](https://busybox.net/) you will need to have support for zstd in [libarchive](https://git.busybox.net/busybox/tree/archival/libarchive) which, despite a proposal from 2021 (facebook/zstd#2806) still hasn't happened yet.

As long as zstd decompression has not formally be added to busybox, or unless you can point me to a _must have_ use case (e.g. **public** images from non-niche projects that use zstd compressed images), I can't see myself adding zstd compression to Rufus.

So, since BusyBox is apparently not moving on adding zstd compression despite getting patches to do so, do you have links to **public** images that use zstd compression?

SplitGemini, 2024-10-19:
There are already a lot of Yocto image I'm currently using that use zstd compression, but in general terms, if you're using libarchive as a compression library, maybe it's not the right time to integrate zstd yet, unfortunately.

pbatard, 2024-11-22:
Okay, since I have just added zstd compression support in Bled, based on the original 2021 patch (which still required some time consuming modifications) I will be reopening this issue so that I can close it in the commit that adds zstd in Rufus.

closed 2024-11-22 by pbatard

pbatard, 2024-11-22:
Note that if you want to test this, you can find Rufus executable with zstd support in the artifacts archive from https://github.com/pbatard/rufus/actions/runs/11981708130.

commit a0d11b35 (2024-11-22): [cmd] add ztsd image compression support

* Based on the latest Bled, which adds ztsd compression support.
* Note that initial extraction of the 512 bytes MBR is very slow, because is seems
  clear that ZSTD was never designed for fast init or processing small elements of
  data, but instead for post init large volume streaming.
* Also note that this code adds 400 KB to the Rufus executable *AFTER UPX COMPRESSION*!
  Hopefully, the BusyBox folks can come up with a better and smaller way to add zstd
  support, because it's clear that the method used by the current BusyBox proposal,
  which is t [... cut]
commit cef7a5cb (2024-11-22): [cmp] add ztsd image compression support

* Based on the latest Bled, which adds ztsd compression support.
* Note that initial extraction of the 512 bytes MBR is very slow, because is seems
  clear that ZSTD was never designed for fast init or processing small elements of
  data, but instead for post init large volume streaming.
* Also note that this code adds 400 KB to the Rufus executable *AFTER UPX COMPRESSION*!
  Hopefully, the BusyBox folks can come up with a better and smaller way to add zstd
  support, because it's clear that the method used by the current BusyBox proposal,
  which is t [... cut]