pbatard/rufus #2593: Unable to extract iso file after update to windows 11 24h2

Maximaz1212, 2024-10-21:
﻿

Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<After windows 11 update to 24h2, when ever i create windows iso bootable usb , it always failed to format the device, being swap USB port to change other thumbdrive to ssd. I'm using Rufus 4.5 and 4.6 Beta, but same problem still happen. Below is my Log.i have check the storage and it in good condition. i can format the usb under windows without a problem.>

Log
Rufus x64 v4.6.2205 (Beta)
Windows version: Windows 11 Pro x64 (Build 26100.2033)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 revoked UEFI bootloaders from embedded list
Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Device eliminated because it was detected as a Hard Drive (score 21 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.0) device 'ADATA SU 650 UAS Device' (7825:A2A4)
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
Device eliminated because it was detected as a Hard Drive (score 8 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Found UAS (USB 3.0) device 'ADATA SU 650 UAS Device' (7825:A2A4)
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
3 devices found
No volume information for drive 0x83
Disk type: FIXED, Disk size: 240 GB, Sector size: 512 bytes
Cylinders: 29185, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {9F791E9E-833F-4858-A57E-B9C2C07D2B1F}
Max parts: 128, Start Offset: 17408, Usable = 240057375232 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {B97C8848-EB95-4C2F-8045-3FED467ECDAB}
  Size: 223.6 GB (240056303616 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
libcdio: Auto-expanding the size of 1-Boot-NoEmul.img
  Image is an ISO9660 image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ACRONIS_MEDIA'
  Size: 2.6 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
Using image: AcronisBootablePEMedia.iso (1.3 GB)
libcdio: Auto-expanding the size of 1-Boot-NoEmul.img
Could not get ISO-9660 file information for file /efi/boot/bootx64.efi
Warning: Failed to extract '/efi/boot/bootx64.efi' to check for UEFI revocation

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 223.6 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{18cb7168-3b06-4892-ac2c-feabf1fd0dff} for exclusive write access
Size : 223.6 GB 468859968 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 18f4:3353
248 Reserved sectors, 57220 sectors per FAT, 2 FATs
7324145 Total clusters
7324144 Free clusters
Clearing out 114752 sectors for reserved sectors, FATs and root cluster...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
write_sectors: Write error [0x00000015] The device is not ready.
  StartSector: 0x00000000, nSectors: 0x80, SectorSize: 0x200
Error clearing reserved sectors
Format error: Write error.
Re-mounted volume as G: after error

Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Found UAS (USB 3.0) device 'ADATA SU 650 UAS Device' (7825:A2A4)
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
3 devices found
No volume information for drive 0x83
Disk type: FIXED, Disk size: 240 GB, Sector size: 512 bytes
Cylinders: 29185, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {EAA4CC98-DDC4-4A36-92A8-29EB3C455A99}
Max parts: 128, Start Offset: 17408, Usable = 240057375232 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {18CB7168-3B06-4892-AC2C-FEABF1FD0DFF}
  Size: 223.6 GB (240056303616 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Found UAS (USB 3.0) device 'ADATA SU 650 UAS Device' (7825:A2A4)
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
3 devices found
No volume information for drive 0x83
Disk type: FIXED, Disk size: 240 GB, Sector size: 512 bytes
Cylinders: 29185, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {EAA4CC98-DDC4-4A36-92A8-29EB3C455A99}
Max parts: 128, Start Offset: 17408, Usable = 240057375232 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {18CB7168-3B06-4892-AC2C-FEABF1FD0DFF}
  Size: 223.6 GB (240056303616 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
2 devices found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 124519, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8AA5EDF9-1EDD-4A61-9412-79EB4B91D9E0}
Max parts: 128, Start Offset: 17408, Usable = 1024209508864 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Detected File System: NTFS
  ID: {D6B43689-3111-BC4B-9B1A-383D7E7F207A}
  Size: 953.9 GB (1024209505280 bytes)
  Start Sector: 40, Attributes: 0x0000000000000000
Found UAS (USB 3.1) device 'IMPERION -1TB UAS Device' (174C:2362)
Found UAS (USB 3.1) device 'Realtek RTL9210B-CG UAS Device' (0BDA:9210)
2 devices found
Disk type: FIXED, Disk size: 1TB, Sector size: 512 bytes
Cylinders: 124519, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8AA5EDF9-1EDD-4A61-9412-79EB4B91D9E0}
Max parts: 128, Start Offset: 17408, Usable = 1024209508864 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Detected File System: NTFS
  ID: {D6B43689-3111-BC4B-9B1A-383D7E7F207A}
  Size: 953.9 GB (1024209505280 bytes)
  Start Sector: 40, Attributes: 0x0000000000000000

```

```

pbatard, 2024-10-21:
Many people are using Rufus with 24H2 and you are the only person reporting this issue, so your problem has to be environmental.

My guess is that you enabled some security settings (or those were enabled for you), that interfere with Rufus accessing the drive, as a `[0x00000015] The device is not ready` error means that something external to Rufus is hindering access. And yes, if it's some kind of Windows security setting, you might be able to format the drive using Windows native tools, which would of course be whitelisted, but not using third party utilities like Rufus.

You may also want to triple check that your USB connection is not flaky.

Considering that you are the only one reporting this issue (for the record, I am personally running 24H2 and have not once ran into this error), I am again pretty confident that it is 100% due to your environment, and that there is nothing that Rufus can do to alleviate it (especially as all Rufus does when you run into that error, really, is ask Windows to write a few blocks of data on the USB &mdash; There's absolutely nothing fancy going on behind the scenes).

Maximaz1212, 2024-10-21:
If that the case, then i also out of idea what kind of app that prevent rufus to access it, since it happen only after updating to 24h2, So reformat the laptop is the only way for me. Thank for the input ya. 

Cheers .

pbatard, 2024-10-30:
Since this issue is clearly environmental and not something anybody else appears to be observing, I will close it.

closed 2024-10-30 by pbatard