pbatard/rufus #2595: FAQ update regarding windows 11 24H2

tmtm24680135, 2024-10-21:
﻿

Checklist
---------
- [ ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, 24H2
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
It seems that version 4.6 has added support for Windows 11 24H2, but the FAQ (https://github.com/pbatard/rufus/wiki/FAQ#this-pc-doesnt-currenty-meet-windows-11-system-requirements) still implies it is not supported.  If 4.6 indeed supports Windows 11 24H2, maybe that FAQ entry should indicate that release 4.6 is required for Windows 11 24H2 and support for Windows 11 releases after that are TBD.

Log
---
```
<FULL LOG>
```

pbatard, 2024-10-21:
As @pineapple63 explained, it just appears that you misread the FAQ entry and interpreted it to mean that anybody who tried to upgrade to 22H4 using Rufus 4.5 will see the error dialog, whereas this is an entry for people who might still get an error, with the error dialog, when using Rufus 4.6.

The FAQ entry makes no condition about the version of Rufus, and simply provides info about what it means, **regardless of the version of Rufus you used to perform the upgrade**, if you see the error.

closed 2024-10-21 by pbatard

tmtm24680135, 2024-10-21:
I definitely did not realize **windows 11 24H2 has introduced new requirements that cannot be by passed** - maybe the FAQ entry title can be changed to reflect this to reduce confusions like mine.

The commit (https://github.com/pbatard/rufus/commit/98a42a235f67f6e535b6ab07c23bd4eb5f8ba770) included in version 4.6 did help forming my tunnel vision.  I am not sure I was seeing the PopCnt and SSE4.2 checks thus would have to test version 4.6 with W11 24H2 on my machine.  Thanks for all the great work.