pbatard/rufus #2601: Rufus problem 2 partitions?

Lisa-rufus, 2024-10-26:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Yesterday i tried to use rufus as intendeed and worked ,my only problem was that i made a mistake on the config the first time and got an error message, when i corrected my mistake and tried running rufus the usb stopped working, now the usb has 2 partitions that i cant access unlees i erase one partition and create a new volume, i dont have the old logs but i have the logs of what happens when i try running rufus, i cant undo the changes on the usb with the windows tool, and i dont know if this kind of problems are becouse rufus  >

Log
---Rufus x64 v4.6.2208
Windows version: Windows 10 Home x64 (Build 19045.5011)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
Found USB 2.0 device 'Generic Storage Device USB Device' (048D:1336)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x02F2E2AB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.0 GB (16105967616 bytes)
  Start Sector: 128, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Detected File System: (Unrecognized)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 31457160, Boot: No
Found 5 additional UEFI revocation filters from remote SBAT
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'WinterOSRev13 W10 LTSC2021x32'
  Detected: Windows 10 ISO (Build 19044.4780)
  Size: 3.0 GB (Projected)
  Note: File on disk is larger than reported ISO size by 572 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: WinterOSRev13 W10 LTSC2021x32.iso (3.0 GB)
Selected Windows User Experience options:
• Disable data collection

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 15.0 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000032] (NB: This system was unable to provide an English error message) - Retrying...
[0x00000032] (NB: This system was unable to provide an English error message) - Retrying...
[0x00000032] (NB: This system was unable to provide an English error message) - Retrying...
[0x00000032] (NB: This system was unable to provide an English error message) - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as F: after error

Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
Found USB 2.0 device 'Generic Storage Device USB Device' (048D:1336)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x030410CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.0 GB (16105062400 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
Found USB 2.0 device 'Generic Storage Device USB Device' (048D:1336)
Device eliminated because it appears to contain no media
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x030410CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.0 GB (16105062400 bytes)
  Start Sector: 2048, Boot: Yes

```
<FULL LOG>
[rufus.log](https://github.com/user-attachments/files/17531405/rufus.log)

pbatard, 2024-10-26:
1. Rufus created the extra partition because you chose to enable _Add fixes for old BIOSes (**extra partition**, align, etc.)_.
2. Your drive appears to have fail independently of what you were doing, which is why you can't reformat/repartition it. This is common with flash drives, because flash memory is **A LOT** more unreliable than people think (and when that happens, a lot of drive internal controllers will switch to read-only mode so that you get a chance to access your data). I'm afraid you will need to replace that drive. And in case you might be under the impression that software like Rufus has the capability to damage flash drives (it doesn't), I also invite you to read [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive).

closed 2024-10-26 by pbatard

Lisa-rufus, 2024-10-27:
Thanks, i already fixed the usb, it can be written and read i just cant use
rufus to make it bootable.