pbatard/rufus #2602: Cannot select install version of win11. It becomes by default home edition. I need professional

trcrsired, 2024-10-26:
why?

thecatontheceiling, 2024-10-26:
Edition upgrading works as intended. If you would elaborate on what's "broken" I'd appreciate it a lot

trcrsired, 2024-10-26:
> The windows installer will automatically select the edition of Windows that matches the product key in the firmware A workaround will never get added to Rufus, as it falls into the catagory of things that can be done post install (even though the edition upgrade mechanism in Windows is a bit broken, athough as to how much of that is down to people trying to use OEM keys to upgrade the edition, i am not sure of), and most people are unlikely to need to install Pro on a device that came with Home

but i do have a windows 10 professional license on my surface pro 6. I installed archlinux on it then come back to windows

trcrsired, 2024-10-26:
https://support.microsoft.com/en-us/surface/surface-pro-6-specs-and-features-ade5cfc2-e99a-6fd1-abbe-c0e8a8a3942d

in fact the professional license comes from microsoft itself

pbatard, 2024-10-26:
Licensing and activation are 100% handled by Microsoft **alone**. Rufus has absolutely nothing to do with it.

As such, I will close this issue.

closed 2024-10-26 by pbatard

thecatontheceiling, 2024-10-27:
You may have a digital license but the Windows installer only cares about the key integrated within the hardware, it doesn't try to check what other licenses you may have gotten from other sources.