pbatard/rufus #2603: Some UI

rdp, 2024-10-30:
﻿

Checklist
---------
- [x ] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Just some feedback on the UI from "fresh user eyes" as it were.  Feel free to close this but here are some ideas...overall it's pretty good, thank you.

1) After the format completes it goes back to saying "ready".  This is actually a bit confusing, I'd expect a prompt saying "success!" or something like that in the UI some how...it's surprising.

Also a way to eject the media after formatting would save some time, as after formatting it I'm not sure if I can immediately pull it or not.

2) the section "persistent partition size" is a bit hard to understand..Is this a partition that survives the next format?  The verbiage is a bit confusing.  Suggestion: something like "allow writing to new media by adding a persistent partition" or something.  Or hide it under advanced drive properties... :)

Cheers and thanks for Rufus, saves so many people!

rdp, 2024-10-30:
Also somewhat unrelated, would be nice if rufus saved its log files after each run.  If I run into a problem with a USB drive using it an hour later, its window might not still be up (or me aware enough that I wanted to go back and look at an old log after retrying it several times).  Basically, if I can't repro an issue, I can't go back to look at my original log for it, either...thanks!

pbatard, 2024-10-30:
1. Rufus displays a very noticeable progress bar that will only go to 100% on success, and turn red on error. If that is not intuitive enough for someone, then I'm afraid no amount of tweaking the status message will do any good. READY is common status indicating that any previous operation has successfully completed and that the application is ready for more work. If you read "READY" as _"Hmmm, I know it says 'ready', but maybe there was an error that wasn't reported somehow...", the problem is with you.
2. Eject after formatting has been asked before. It's not going to happen because if you need it, it's literally a click on the desktop. There is no point with replicating an operation that is super readily accessible on Windows, and the thing is, if I were to add such a feature, then the people who want to copy extra data to a drive after creation (e.g. Linux or Windows Wifi Drivers) would then ask for ways to control whether Rufus should eject the drive or not, and you now have created an unwarranted configuration menu again, for a feature that is literally always one click away on the Windows desktop.
3. If you have trouble understanding what persistent means, then ask the Linux folks to change their terminology. I did not invent the terms "persistent" or "persistence" and I'm sorry, but it should be obvious that this feature is about persistence for live Linux drives (which means that, no, it'll not be preserved if you reformat the drive). If you are confused, a simple Google search with clarify that, and please bear in mind that Rufus is **not** and will never be an application that is designed to hold your hand, on account that installing an OS or even simply booting a drive are nontrivial operations that you are supposed to familiarize yourself with. I am actually **intentionally** using terminology that will not be familiar to non tech-savvy folks so that they are (hopefully) scared away from using a utility design to accomplish a task that they are likely to have trouble with if they don't do their research first.
4. All versions of Rufus save the most recent log after usage by default. Check your `%LOCALAPPDATA%\Rufus\` folder (knowing that `%LOCALAPPDATA%` should translate to `C:\Users\YOUR USER NAME\AppData\Local\`. And yes, you can type `%LOCALAPPDATA%` in File Explorer and Windows will happily resolve it). You can also make the logs persistent (meaning log will be appended for each session) by pressing <kbd>Ctrl</kbd>-<kbd>P</kbd>. This is [documented in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#power-keyscheat-modes).

As I believe I have answered your questions, I will close this issue.

closed 2024-10-30 by pbatard

rdp, 2024-12-13:
1. It isn't a big problem, since it says "ready" and I suppose you can "guess" from the absence of a failure message that it worked.  The normal workflow for me using Rufus is to start a "burn" then move on to something else while it's writing, so the only progress bar I see is the on in the taskbar, so it's hard to make sure it went to 100...not a hugie...
2.  It's several clicks to find it from the taskbar :)
3. Current way "obfuscation by confusing terminology" is OK...even after our conversation here I don't understand what it does fully LOL.  Up to you.
4. Normal behavior after a failure is to "try again" (I did), and it did work the second time.  But then I can't leave error reports as easily.  It wasn't intuitive from the "logs" button that previous logs were available.  Some UI feedback.

Just some feedbacks, feel free to leave closed and ponder, peace and thanks for Rufus!