pbatard/rufus #2607: Buggy AMI NTFS Bootloader

DukeJukemEG, 2024-11-07:
How does one speak to you about your software if there's no discussion page?

Please make an exFAT option so people with the "Buggy AMI NTFS" whatever can burn Windows ISOs that aren't directly from the media creation tool or sites like uupdump. I'd like to put W11 IoT Enterprise LTSC from massgrave on my machine but what do I burn it with since it's like 6 gigs?

Rufus isn't an option because I'm stuck with NTFS as my only option and selecting the "verify checksum" thing will just show me errors no matter how many times I make a usb (thanks asus.) Ventoy isn't an option because 24H2 in general is broken there. isotousb sucks. rufus is also throwing checksum errors on 24h2 isos downloaded form the media creation tool when choosing fat32 and checking the verify checksum stuff as well. So everything is broken I guess.

What do i do?

pbatard, 2024-11-07:
> How does one speak to you about your software if there's no discussion page?

This issue tracker or my e-mail feature prominently on the web site as well as the about dialog of the application.

>  and selecting the "verify checksum" thing will just show me errors no matter how many times I make a usb (thanks asus.)

Not if you boot from the second partition. You will see 2 boot options for the device, and the second one will boot through UEFI:NTFS that will replace the buggy AMI NTFS driver with a working non AMI one.

At any rate, the MD5 checksum error reports do **NOT** mean that the files are corrupted. Just that, when using the AMI NTFS driver (**WHICH WINDOWS DOES NOT USE FOR ITS INSTALLATION PROCESS**) you may read erroneous data.

At any rate, thanks to Microsoft's Secure Boot signing policies, I cannot provide a Secure Boot compatible exFAT driver for UEFI:NTFS, which is why Rufus does and will continue to use NTFS by default (since, again, any report about MD5 errors that are due to the AMI NTFS driver will not impact Windows installation).

So your options are:
1. Boot using the second partition rather than the first (Since I am not aware of any motherboard UEFI firmware that will only propose booting from the first but not the second).
2. Ignore the MD5 errors if you do boot using the first partition, since (thankfully) the Windows installer does not actually read from the media using the buggy AMI NTFS driver (except for the initial bootloader, but it seems the bootloader is too small in size to be affected by the bug).
3. Don't use the MD5 validation (since there is a good reason why it is not enabled by default in Rufus and you cannot trust its results **if** you get the alert about the AMI driver).

closed 2024-11-07 by pbatard

DukeJukemEG, 2024-11-08:
Wow. I was unaware of what the second partition was for on the Rufus USB. I figured it was just a special Rufus thing and now I know it's more special than i thought lol. I feel stupid :/. 

I apologize if I came off rude. After reading the rules I figured my post would be autodeleted since I didn't have a log etc. I have installed Windows and ignored the checksum errors and all seemed well but my own paranoia had me questioning if my install was truly stable. Thanks for clarifying :D.