pbatard/rufus #2614: req new feature: Select\Enable Windows Setup Edition via adding ei.cfg to Source folder

ssarfaty, 2024-11-14:
Can’t Select Windows 11/10 Edition During Clean Installation if a previuse license installed and to change that need to add a "ei.cfg" 

the file should be stored under the "Sources" folder.

file with the following content will cause the setup to always show the user the selection of editions list to choose what to install:
[Channel]
_Default
[VL]
0

Alternatively, you can also automate the edition selection process by configuring the EI.cfg file to install your preferred edition of the OS. For example, if you want Windows Setup to automatically select the Windows 11 Pro edition during installation, use the following format for the EI.cfg file such as this example:
[EditionID]
Professional
[Channel]
_Default
[VL]
0

this is a request for Rufus to add this feature when creating a new image , at least as the #1 example i have posted.

Thanks!

pbatard, 2024-11-14:
`ei.cfg` can be added by the user.

Also, on all the pre-activated Windows 11 machines I used, I have always seen the option to state: "I don't have a license" and then manually select the edition of Windows to install, so my experience is very different from your statement of _"Can’t Select Windows 11/10 Edition During Clean Installation if a previuse license installed"_. So there has to be something else at play in your environment.

At any rate, I see it as too uncommon an operation (switching licenses at installation time), since, unless you either have a very expensive subscription to MSDN (or you are pirating Windows), you are unlikely to be hopping editions, and Windows activation and licensing matters are outside of the realm of what I design Rufus to deal with.

&rarr; Rejected

closed 2024-11-14 by pbatard

Akrotkiv, 2024-12-18:
I just installed Win11 download from microsoft right now. I didn't see any chance to choose from edition. (tried 2 times) Automaticaly selected "HOME" because it was previously on this Laptop. New disk for system was inserted. So it reads from bios/uefi.