pbatard/rufus #2616: Redirects from rufus.akeo.ie Not Working - older versions broken

own3mall, 2024-11-14:
http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys 404 not found

https://rufus.ie/files/syslinux-6.04/ldlinux.sys exists...

Log from Rufus 2.18.1213 on Windows 7:

```
Using image: BookwormPup64_10.0.8.iso (777 MB)
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/20200816/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.04/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
```

pbatard, 2024-11-14:
That's expected. This will never be fixed because I would have to release a new version of Rufus for Windows 7, which will [NEVER, EVER HAPPEN](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED).

If you want support with Rufus, please use Windows 10 or later.

closed 2024-11-14 by pbatard

own3mall, 2024-11-14:
Or you could fix your web redirects properly if you know anything about web servers.  How embarrassing!

pbatard, 2024-11-14:
> all it would take (...)

Yeah, sure, all it would take.

rufus.akeo.ie is hosted **by GitHub** (whose webserver configuration I do not have access to) from https://github.com/pbatard/rufus/tree/gh-pages and must redirect to rufus.ie (which is itself served by https://github.com/pbatard/rufus-web, another GitHub pure static hosting) so that people visiting older links do get redirected to the new URL. Oh and, for obvious reasons, I want an HTTP to HTTPS redirect on rufus.ie.

And, before you ask, no I am not going to duplicate content between 2 GitHub static hosts (even with an automated GitHub Actions script in the background) just to keep a dwindling bunch of entitled Windows 7 and XP users happy.

So, if you have a solution I can apply to https://github.com/pbatard/rufus/tree/gh-pages and https://github.com/pbatard/rufus-web that'll make your "all it would take" solution works in the context of GitHub static hosting and limited CNAME handling, and that doesn't require duplicating the content, I'm all ears. Otherwise, you can take your annoyance at people not wanting to jump through a lot of hoops, and investing a lot of time trying to support OBSOLETE platforms, somewhere else.

own3mall, 2024-11-14:
Oh, I'm sorry.  I didn't realize you hand cuffed yourself to being dependent on Microsoft and GitHub.  That's why it's always better to do as much as you can yourself than rely on third parties.  Thanks for Rufus, but it really is a shame that a lot of developers have become so lazy and arrogant when it comes to making their software work on as many platforms as possible.  It should not be up to the developer to limit what his software works on, or to tell me what I'm running is obsolete.  That is not for you to judge.  Target XP in your development  and it should work on anything that comes after unless MS does some dumb stuff to break it, which they probably will.  No one cares anymore, so yeah, I guess that's that... Just keep writing more and more inefficient software while more e-waste is generated for no good reason... It's a shame we are short sited and believe "security" is the number one priority over functional and working software...

pbatard, 2024-11-14:
>  I didn't realize you hand cuffed yourself to being dependent on Microsoft and GitHub.

I used to host the Rufus content on a dedicated VPS. Until a user, probably not that dissimilar to you, took objection to my approach to software development and thought they could teach me a lesson by DDoSing that server... which helped me decide, that I might as well make use of the contingency plan I had for web server hopping, and let GitHub handle the DDoS in order to concentrate on what really matters (which is my software running well on **current** platforms). I haven't looked back since.

> Target XP in your development and it should work on anything that comes after

LOL. Yeah, great. So that means no Windows 11 customisation for Rufus users and especially no TPM bypass, unless I write my own ISO mounting driver. And no Windows To Go. And no VDS facility. And the constant agony of having to adjust the layout and localization so that no platform visibly feels left behind. Oh and please don't remind me how GPT and USB 3.0 support were an absolute blast on Windows 7 and XP and how it took the Windows 10 Creators Update to finally have the vast majority of USB Flash Drives be able to mount multiple partitions...

If you think Windows does not evolve between major versions, and does not bring welcome changes to both users and software developers, then I guess there's a lot I could say too about short-sightedness...

The software might be free. My time isn't. I don't have the luxury of thinking that my software is "good enough" so that it doesn't need to evolve to cater to current needs (that did not exist at the time of XP and 7) or that I will have all the time in the world to **properly** test and ensure that Rufus works well on 6 different versions of Windows (XP, Vista, 7, 8, 10, 11).

IMO, what really is a shame is that people continue to believe that everything they haven't been closely involved with can be summarized to an easily digestible fallacy, such as that something that runs on XP will be more than good enough for later versions of Windows, and does not need much time investment.

own3mall, 2024-11-14:
I understand that it's not easy.  It sucks.  I'm a developer too.  It's hard to support everything, especially as new stuff is released, and a lot of times, Microsoft does make things easier for a developer, but by doing so, they drop support for older operating systems.  So, while it may be easier for the developer to use this new easy solution, realize that there are trade-offs.  But, there are always some things that can be done to prevent things like this from happening.  Such as, not moving to a new domain.  Why didn't you just stay on `rufus.akeo.ie` rather than have `rufus.ie`?  Was that necessary?    

https://github.com/pbatard/rufus/commit/e6d8f25bf1ca9fcda32edebfcb498e1a76cb4f6f

Looks like if I can find the rufus-2.18 source, it would be an easy update and and re-build.  Do you obfuscate your strings somehow?  A patcher utility I wrote is not able to find `rufus.akeo.ie` as a string value to replace with `rufus.ie` and null bytes in the compiled exe...

pbatard, 2024-11-14:
Feel free to fork the project and release and maintain your own versions of Rufus for Windows XP.

You do realize that if I were to produce a new version of Rufus that is compatible with XP, it would imply that I am still supporting XP. Which I don't.

So, I will reiterate what I say to users who insist on running Rufus on Windows XP or Windows 7: You are 100% on your own.

own3mall, 2024-11-15:
Mkay, there you go:

https://github.com/own3mall/rufus/releases/tag/2.18

https://github.com/own3mall/rufus/commits/master/

Seems to be working now :+1:

own3mall, 2024-11-16:
Oh, and I also compiled an XP version of Rufus 2.18 that actually works on XP too --- cool!

https://github.com/own3mall/rufus/releases/tag/2.18

own3mall, 2024-11-18:
@pbatard Building and compiling the source code clearly involves more than changing the platform toolset for the projects to v141_xp.  Could you please provide instructions as to how to properly compile / build it for XP?  I'm not able to build a fully functional version of 2.18 on XP using the 2.18 source code just opening the solution file in Visual Studio 2017 and building.  Also, your releases are under 1 MB.  The built versions in VS are over 2.5 MB.  How does one build and compile the source code in the same manner you do?

pbatard, 2024-11-18:
> How does one build and compile the source code in the same manner you do?

By using MinGW and invoking `make release`.

Which is about as far as I'm going to help you compile Rufus for XP. Everything is public, but you might have to look for it, and I have no more time to spend trying to help people with platforms that have been obsoleted.

If you need more help than this, then I will reiterate what I stated above (and this time I mean it): You are 100% on your own.

own3mall, 2024-11-18:
@pbatard Thank you.  I got the build working with MinGW which was quite complicated to setup in Windows.

Visual Studio does create a working exe (that's not compressed - since no upx package), but there's libcdio issues in the Visual Studio compiled exe that aren't present in the build created by MinGW.  I wouldn't advertise it as working with Visual Studio... 

https://github.com/own3mall/rufus/commit/e1a4fe2d3c9faf9730dd554caea31c95bdc435e2

I also created an http mirror to pull from your https mirror so that XP can work.