pbatard/rufus #2619: Issue when Mount WIndowsToGo using Custom Windows 11 ISO

pitolu1212, 2024-11-17:
Hello Friend sorry for my last post without upload the LOG.

I can mount my Customs Windows 11 with rufus in Standard WIndows Installation with any problem but when try mount the same custom ISO in WindosToGo mode trip the Message: Unable Patch/Setup files for Boot.

It seems to be a problem with the unattend.xml file im not sure!!

[rufus.log](https://github.com/user-attachments/files/17790915/rufus.log)

pitolu1212, 2024-11-17:
Thank you for fast reply. 
then Rufus is not compatible for mount custom windows ISO in wintogo mode? because ussing wintohdd software is possible without problem!!

pbatard, 2024-11-17:
Yup. Rufus does not support custom Windows ISOs that already have an `unattend.xml`. If you want to use such an ISO, you will have to customize the installation drive on your own.

closed 2024-11-17 by pbatard

pitolu1212, 2024-11-17:
I understand, thank you for your time.

pbatard, 2024-11-17:
Note that, if you still want to use a custom ISO with an existing `unattend.xml` in Rufus, you should make sure that you uncheck ALL the options from the Windows User Experience dialog. The conflict exists because Rufus will attempt to create an `unattend.xml` if you use these options, so if you uncheck them all, the process should complete (but of course, without extra customization, which I guess is what you were hoping to achieve).

pitolu1212, 2024-11-17:
Thank you soo much Friend! I try mount again the ISO without a file unatend.xml included.

pitolu1212, 2024-11-21:
Hello Friend. I tried again mount custom win11 ISO in wintogo mode with all Windows User Experience Dialog options uncheck,but the result its same. can not be completed.

pbatard, 2024-11-21:
Well, you will just have to make sure that there is no `unattend.xml` on your custom ISO. I'm afraid I simply cannot support Windows custom ISOs, as there are simply too many ways they can interfere with Rufus.

pitolu1212, 2024-11-21:
Of course friend, my ISO included unattend.xml in windows/panther/ but in your previous reply, you said if uncheck All options in windows user experience dialog not have any problem, but the conflict persist.

pbatard, 2024-11-21:
I said "should". Precisely because I thought it was worth a try, but I couldn't make any assumption that it would work. I never made any promise that it would and, once again, I simply don't have the scope to support custom Windows ISOs in Rufus, so if they work, great, but if they don't, then I will rely on you to alter them in a manner that makes them work with Rufus.