pbatard/rufus #2620: Feature Request: Support zstandard compression

benfi5, 2024-11-19:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
As a manufacturer of embedded devices we are using zstandard (.zst) as compression for our linux (.wic) images because of the impressive performance. 

It would be handy for a user to flash it directly to an sd-card, instead of first extract it by using another tool, like 7zip >= 24.7.

- Rufus v4.2.2074 does not detect this compression method.
- "Image does not have a Boot Marker" Log entry: This is not the problem. Rufus works, when the .wic.zst file is uncompressed first.  

Reference: (https://github.com/facebook/zstd/blob/dev/doc/zstd_compression_format.md#zstandard-compression-format)

Log
---
```
Rufus x64 v4.2.2074 (Portable)
Windows version: Windows 10 Enterprise x64 (Build 19045.5011)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found 517 officially revoked UEFI bootloaders from embedded list
Found 0 additional revoked UEFI bootloaders from this system's SKUSiPolicy.p7b
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 6
Disk GUID: {00000000-0000-0000-0000-000065E5E543}
Max parts: 128, Start Offset: 17408, Usable = 31914949120 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot_sd'
  Detected File System: FAT16
  ID: {9BD9911C-9308-4702-83B8-41CB3B8FAC36}
  Size: 10 MB (10485760 bytes)
  Start Sector: 10240, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'primary'
  Detected File System: ext4
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000011}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 30720, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'primary'
  Detected File System: (Unrecognized)
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000012}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 3176448, Attributes: 0x0000000000000000
Partition 4:
  Type: Linux Data Partition
  Name: 'data_sd'
  Detected File System: ext4
  ID: {6C50EA0F-6AA8-4970-9842-9CF43287BB2E}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6322176, Attributes: 0x0000000000000000
Partition 5:
  Type: Linux Data Partition
  Name: 'log_sd'
  Detected File System: ext4
  ID: {78C1506D-F561-4A33-B150-85CFD504986C}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6453248, Attributes: 0x0000000000000000
Partition 6:
  Type: Linux Data Partition
  Name: 'docker_sd'
  Detected File System: ext4
  ID: {8ED0CFDC-1FB6-4382-9DDC-537BEEDDCB45}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6584320, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  'C:\debug-image-debian-bookworm-compact-v3-sd-2.3.0-alpha.1+34+3ce1312.wic.zst' doesn't look like an ISO image
Disk image analysis:
  Image does not have a Boot Marker
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 6
Disk GUID: {00000000-0000-0000-0000-000065E5E543}
Max parts: 128, Start Offset: 17408, Usable = 31914949120 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot_sd'
  Detected File System: FAT16
  ID: {9BD9911C-9308-4702-83B8-41CB3B8FAC36}
  Size: 10 MB (10485760 bytes)
  Start Sector: 10240, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'primary'
  Detected File System: ext4
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000011}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 30720, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'primary'
  Detected File System: (Unrecognized)
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000012}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 3176448, Attributes: 0x0000000000000000
Partition 4:
  Type: Linux Data Partition
  Name: 'data_sd'
  Detected File System: ext4
  ID: {6C50EA0F-6AA8-4970-9842-9CF43287BB2E}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6322176, Attributes: 0x0000000000000000
Partition 5:
  Type: Linux Data Partition
  Name: 'log_sd'
  Detected File System: ext4
  ID: {78C1506D-F561-4A33-B150-85CFD504986C}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6453248, Attributes: 0x0000000000000000
Partition 6:
  Type: Linux Data Partition
  Name: 'docker_sd'
  Detected File System: ext4
  ID: {8ED0CFDC-1FB6-4382-9DDC-537BEEDDCB45}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6584320, Attributes: 0x0000000000000000
Failed to scan image
```

pbatard, 2024-11-19:
Duplicate of #2590. See the answer there. I'm also going to ask the same thing I asked there: do you have links to **public** images that use zstd compression?

pbatard, 2024-11-22:
Okay, since I have just added zstd compression support in Bled, based on the original 2021 patch (which still required some time consuming modifications) I will be reopening this issue so that I can close it in the commit that adds zstd in Rufus.

closed 2024-11-22 by pbatard

pbatard, 2024-11-22:
Note that if you want to test this, you can find Rufus executable with zstd support in the artifacts archive from https://github.com/pbatard/rufus/actions/runs/11981708130.

benfi5, 2024-11-23:
So great! I just tested it. Flashing the image to sd-card and booting afterwards from it worked well! Many thanks for the immediate action.

Here the log for your reference:

Rufus x64 v4.7.2211 (Alpha)
Windows version: Windows 10 Enterprise x64 (Build 19045.5131)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Killing stuck thread!
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 revoked UEFI bootloaders from embedded list
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 983, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {00000000-0000-0000-0000-000065E5E543}
Max parts: 128, Start Offset: 17408, Usable = 8090778112 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot_rescue'
  Detected File System: FAT32
  ID: {82FC67AB-CB4B-4156-AECF-DF7D7E453127}
  Size: 652.8 MB (684561408 bytes)
  Start Sector: 10240, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'root_rescue'
  Detected File System: ext4
  ID: {AA911FF2-7068-4F78-AA9F-D6CFE67EE911}
  Size: 6.9 GB (7399800832 bytes)
  Start Sector: 1347584, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  'C:\artifacts\images\c6c\keb-debug-image-debian-bookworm-compact-v3-sd-2.4.0-PullRequest6856.18+a1dce57.wic.zst' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: keb-debug-image-debian-bookworm-compact-v3-sd-2.4.0-PullRequest6856.18+a1dce57.wic.zst (186.5 MB)
Timeout while retrieving conflicting process list

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
Writing compressed image:
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Closed Windows format prompt
Opened \\.\PhysicalDrive1 for shared write access
Closed Windows format prompt
Remounted \\?\Volume{71df5db5-a7cd-11ef-8e25-900f0c460a8a}\ as D:

Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 983, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 6
Disk GUID: {00000000-0000-0000-0000-000065E5E543}
Max parts: 128, Start Offset: 17408, Usable = 8090778112 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'boot_sd'
  Detected File System: FAT16
  ID: {103E54DC-E1E4-4386-9B93-CB051276FA93}
  Size: 10 MB (10485760 bytes)
  Start Sector: 10240, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'primary'
  Detected File System: ext4
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000011}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 30720, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'primary'
  Detected File System: (Unrecognized)
  ID: {FEDCBA98-7654-3210-CAFE-5E0710000012}
  Size: 1.5 GB (1610612736 bytes)
  Start Sector: 3176448, Attributes: 0x0000000000000000
Partition 4:
  Type: Linux Data Partition
  Name: 'data_sd'
  Detected File System: ext4
  ID: {3B0CA318-FE58-4085-BD48-CA122315D527}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6322176, Attributes: 0x0000000000000000
Partition 5:
  Type: Linux Data Partition
  Name: 'log_sd'
  Detected File System: ext4
  ID: {5654FBEF-DBAF-4862-8D80-4D77478E510E}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6453248, Attributes: 0x0000000000000000
Partition 6:
  Type: Linux Data Partition
  Name: 'docker_sd'
  Detected File System: ext4
  ID: {DE269289-95F2-4288-87F0-4730D40ED415}
  Size: 64 MB (67108864 bytes)
  Start Sector: 6584320, Attributes: 0x0000000000000000

commit a0d11b35 (2024-11-22): [cmd] add ztsd image compression support

* Based on the latest Bled, which adds ztsd compression support.
* Note that initial extraction of the 512 bytes MBR is very slow, because is seems
  clear that ZSTD was never designed for fast init or processing small elements of
  data, but instead for post init large volume streaming.
* Also note that this code adds 400 KB to the Rufus executable *AFTER UPX COMPRESSION*!
  Hopefully, the BusyBox folks can come up with a better and smaller way to add zstd
  support, because it's clear that the method used by the current BusyBox proposal,
  which is t [... cut]
commit cef7a5cb (2024-11-22): [cmp] add ztsd image compression support

* Based on the latest Bled, which adds ztsd compression support.
* Note that initial extraction of the 512 bytes MBR is very slow, because is seems
  clear that ZSTD was never designed for fast init or processing small elements of
  data, but instead for post init large volume streaming.
* Also note that this code adds 400 KB to the Rufus executable *AFTER UPX COMPRESSION*!
  Hopefully, the BusyBox folks can come up with a better and smaller way to add zstd
  support, because it's clear that the method used by the current BusyBox proposal,
  which is t [... cut]