pbatard/rufus #2622: winget version of Rufus creates rufus.com file.  Gives error: "Unable to synchronize with GUI application." on second run.

NBKRedSpy, 2024-11-21:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When using the winget version of rufus, it creates a rufus.com file in the current directory.  After closing rufus, the file remains.  If the user tries to run rufus again in the same directory, it will give the error:

"Unable to synchronize with GUI application."

Reproduction steps are after the log.

There are two solutions:
* Use the version from github
* Delete rufus.com after every run.

**Note**:  The log is from the first run (the actual rufus.exe).  There is no log from the invalid rufus.com execution.

Log
---
```
Rufus x64 v4.6.2208
Windows version: Windows 10 Pro x64 (Build 19045.5131)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3740, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {C8C78BF9-583B-4B64-AEB2-EAF5FEEF7EED}
Max parts: 128, Start Offset: 17408, Usable = 30765185536 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {9542757D-59D0-4F39-AF5A-69133A6CB272}
  Size: 28.7 GB (30763061248 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {69358492-79E0-4CDC-84E5-09FF8F1E4F4B}
  Size: 1 MB (1064448 bytes)
  Start Sector: 60086187, Attributes: 0x9000000000000000
Found 5 additional UEFI revocation filters from remote SBAT

```

# Reproduction
* Use winget to install rufus:

`winget install rufus.rufus`

* Open a command prompt.
* Run `rufus`
* close rufus
* Note:  There is now a rufus.com in the current directory.
* Run `rufus`
* The dropped rufus.com is run instead of the real rufus.exe and gives the error:  
`Unable to synchronize with GUI application.`

# Additional Information
The winget installed rufus.exe's hash is the same as the github version's hash.  
Running the github version of rufus doesn't have this issue.

Winget info:
```
Found Rufus [Rufus.Rufus]
Version: 4.6
Publisher: pbatard
Publisher Url: https://github.com/pbatard
Publisher Support Url: https://github.com/pbatard/rufus/issues
Author: pbatard
Moniker: rufus
Description: Rufus is a utility that helps format and create bootable USB flash drives.
Homepage: https://rufus.ie/
License: GPL-3.0
License Url: https://github.com/pbatard/rufus/blob/HEAD/LICENSE.txt
Copyright: Copyright  ┬⌐ 2011-2023 Pete Batard and Rufus contributors
Release Notes:
  - Add a new setup.exe wrapper to bypass Windows 11 24H2 in-place upgrade restrictions (#2568)
  - Add TimeZone to regional options replication (#2499)
  - Set local account passwords to not expire by default
  - Fix an error when trying to write compressed VHD images (#2468)
  - Fix an error when invoking Rufus from the PowerShell commandline (#2556)
  - Improve revoked UEFI bootloaders check to support Linux SBAT, Windows SVN and cert DBX
  - Improve support for ReactOS boot media (#2589)
Release Notes Url: https://github.com/pbatard/rufus/releases/tag/v4.6
Tags:
  boot
  formatting
  foss
  uefi
  unetbootin
  usb
  utility
Installer:
  Installer Type: portable
  Installer Url: https://github.com/pbatard/rufus/releases/download/v4.6/rufus-4.6.exe
  Installer SHA256: 8279696c1d78b14618500e9135886a3667b9decc65946f3729002e4bfdbb20ab
  Release Date: 2024-10-21
  Offline Distribution Supported: true
```

The path to the exe installed from winget:
<user profile>\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.exe

hash: 8279696C1D78B14618500E9135886A3667B9DECC65946F3729002E4BFDBB20AB

github version:

rufus-4.6.exe
hash: 8279696C1D78B14618500E9135886A3667B9DECC65946F3729002E4BFDBB20AB

Hash for rufus.com
5F819F6EAE4B5845C082EDF14CB389AB9805BC3C17440F3B5398D4FDD0079FFE

pbatard, 2024-11-21:
> When using the winget version of rufus, it creates a rufus.com file in the current directory.

This is normal from any version that runs from the commandline. This is the "commandline hogger" that is described [here](https://github.com/pbatard/rufus/tree/master/res/hogger) (along with the reasons why we need one).

This file should be deleted when Rufus terminates properly. So if you see this file persist after you closed Rufus, then Rufus did not terminate properly. My testing with the Rufus 4.6 release executable shows that, when running Rufus from commandline and closing Rufus, `rufus.com` does get deleted as expected, so, considering that winget uses the exact same executable, something is happening in your environment that is either preventing Rufus from closing properly, or that is interfering with the deletion of the file (for instance, you may have a security solution that thinks the `rufus.com` is suspicious and might be preventing its deletion, as it's a very common occurrence to have some security solution be paranoid about what Rufus does).

>  If the user tries to run rufus again in the same directory, it will give the error: `Unable to synchronize with GUI application.`

That `Unable to synchronize with GUI application.` [error comes from `rufus.com`](https://github.com/pbatard/rufus/blob/master/res/hogger/hogger.c#L22) (i.e the commandline hogger) and not the actual Rufus application. So you are not running `rufus.exe` but `rufus.com` (which will happen if you just type `rufus` without extension and there is a `rufus.com` and a `rufus.exe`).  For the record, I have also tried renaming `rufus-4.16.exe` to `rufus.exe` and just issuing `rufus` from the commandline and still saw no issue with the `rufus.com` creation and deletion.

So, as far as I am concerned, I see no issue with `rufus.com` when running Rufus from commandline, which leads me to reiterate that it's either an issue with your environment, that either prevents Rufus (GUI) from closing properly or from deleting `rufus.com`, or with whatever the packagers of the winget version do, which I have no involvement with.

Considering that you state that you have no issue when running the GitHub version, you may want to contact the winget maintainers (which again, I have no involvement with) to ask them about this issue.

NBKRedSpy, 2024-11-21:
I'm not sure which app is responsible for the rufus.com file creation and delete, so posting here.

Summary: 
* Since this project does not maintain the winget package, this really just informational I suppose.
* winget puts apps in the AppData, which cannot be written to.
* rufus.exe (hogger?) creates the rufus.com in the current directory.  But oddly also tries to write it to the exe's path multiple times, which fails.
* rufus.exe tries to delete rufus.com multiple times, but only checks the rufus exe's directory and doesn't seem to know that it can be in the working directory instead.

Procmon's output for the related rufus.com create/delete requests from rufus.exe:

| Time of Day         | Process Name | PID  | Operation                | Path                                                                                                                  | Result          | Detail                                                                                                                                                                                      |
|---------------------|--------------|------|--------------------------|-----------------------------------------------------------------------------------------------------------------------|-----------------|---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| 10:30:22.6679824 AM | rufus.exe    | 7232 | QueryNameInformationFile | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.exe | SUCCESS         | Name: \Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.exe                                                                   |
| 10:30:22.6773696 AM | rufus.exe    | 7232 | CreateFile               | C:\work\rufus.com                                                                                                     | SUCCESS         | Desired Access: Generic Read/Write, Disposition: OverwriteIf, Options: Synchronous IO Non-Alert, Non-Directory File, Attributes: N, ShareMode: Read, AllocationSize: 0, OpenResult: Created |
| 10:30:22.6775432 AM | rufus.exe    | 7232 | WriteFile                | C:\work\rufus.com                                                                                                     | SUCCESS         | Offset: 0, Length: 2,048, Priority: Normal                                                                                                                                                  |
| 10:30:22.6776326 AM | rufus.exe    | 7232 | CloseFile                | C:\work\rufus.com                                                                                                     | SUCCESS         |                                                                                                                                                                                             |
| 10:30:28.4439156 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
| 10:30:28.6466564 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
| 10:30:28.8612821 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
| 10:30:29.0633482 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
| 10:30:29.2782548 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
| 10:30:29.4930331 AM | rufus.exe    | 7232 | CreateFile               | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.com | NAME NOT FOUND  | Desired Access: Read Attributes, Delete, Disposition: Open, Options: Non-Directory File, Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a           |
[... cut ...] .Source_8                                                                                         |
| 10:30:30.7460801 AM | rufus.exe    | 7232 | QueryNameInformationFile | C:\Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.exe | SUCCESS         | Name: \Users\SomeUser\AppData\Local\Microsoft\WinGet\Packages\Rufus.Rufus_Microsoft.Winget.Source_8wekyb3d8bbwe\rufus.exe                                                                   |

NBKRedSpy, 2024-11-21:
If the create/delete is part of hogger, I can post a bug on that project's repo.

pbatard, 2024-11-21:
> I'm not sure which app is responsible for the rufus.com file creation and delete

Rufus is.

From your extra info, it appears that Rufus may not properly delete `rufus.com` when it is created in a different directory from the executable, and indeed, testing shows that unless the command prompt is set to the same directory as the Rufus executable, then `rufus.com` will be created but not deleted.

So that would be a Rufus bug, which I am going to look into.

And I guess, if you just type `rufus` from the commandline, then the minute there is a `rufus.com` remaining in your directory, you are going to see the issue.

Not sure about the `rufus.com` calls in the AppData directory, as there really is only [one location in the code, at startup](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L3418-L3429), where we create the `rufus.com` commandline hogger (and this is not code that is part of a loop).

Note however that Microsoft uses the `CreateFile()` API even when trying to open an existing file (or simply checking if a file exists), and not only when actually creating a file. This might be expected checks as to whether the hogger exists and needs to be deleted, and the path being set to AppData might be the same issue that prevents deletion, if we are looking in the wrong directory.

I'll look into this further when I have some time, and keep you updated but by the looks of it, this doesn't appear to be a winget issue, so thanks for looking into it further.

commit 8d1ed44e (2024-12-03): [cmd] fix hogger not being deleted on exit when executable is in a different directory

* Closes #2622.
closed 2024-12-03 by pbatard