pbatard/rufus #2623: Error: Unable to patch/setup files for boot.

vonguyenkhang, 2024-11-22:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
**Error: Unable to patch/setup files for boot.**
[image]
I have rebooted my PC and tried again (3 times).
I'm using Rufus Portable downloaded from https://rufus.ie/
Possibly related issues: https://github.com/pbatard/rufus/issues/1884 and https://github.com/pbatard/rufus/issues/2050

Log
---
```
There was an error creating your Issue: body is too long (maximum is 65536 characters).
```
[rufus.log](https://github.com/user-attachments/files/17856521/rufus.log)

pbatard, 2024-11-22:
Differences between the PC that works and the PC that fails:

The PC that fails is using a nonstandard temp drive that is mapped to `R:`.

Can you explain how you set up your temp drive on the Windows PC that failed?

My guess is that you are you are running into [this issue](https://github.com/pbatard/rufus/blob/fcdde3dcb8db1f31d530470ad3c48f2313570705/src/rufus.c#L3314-L3318) where, if you use a symlink for your temp directory, it breaks Windows mounting facilities, which I've attempted a couple of times to work around, but which is a **major PITA to try to fix**.

I'm afraid that my take on this is that I no longer want to bother with issues that stem from people having deviating from the Windows standard configuration because, as you can imagine, I simply can't spent my days chasing about the many, many ways this can break things.

So here are the two things I can suggest:

1. Please install and run [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview) (as Admin) on the PC that fails, and see if you get additional messages about the temp directory that might be insightful. When Rufus starts, you should see a section like this in DebugView, which may contain additional data or warnings about the temp dir:  
```
[27064] *** Rufus init *** 
[27064] Cur dir: 'C:\Users\USERNAME\Desktop\' 
[27064] App dir: 'C:\Users\USERNAME\Desktop\' 
[27064] Sys dir: 'C:\WINDOWS\system32' 
[27064] Usr dir: 'C:\Users\USERNAME' 
[27064] Dat dir: 'C:\Users\USERNAME\AppData\Local' 
[27064] Tmp dir: 'C:\Users\USERNAME\AppData\Local\Temp\' 
[27064] Binary executable is signed by 'Akeo Consulting' 
```
2. From an administrative command prompt, try to run `dism /cleanup-wim` or, if that doesn't work, try to follow the additional steps from [this post](https://superuser.com/a/1317923/286681).

vonguyenkhang, 2024-11-22:
@pbatard you're correct that on the PC with issue, the Temp folder is not standard but symlinked.
I use this tool [ImDisk Toolkit](https://sourceforge.net/projects/imdisk-toolkit/) to create the RAM drive and set the Temp folder.
[image]
As you can see in the screenshot above, the Temp folder is actually symlinked to `R:\Temp`.
The intriguing parts:
1. The working PC also has the same config (Temp folder also symlinked to `R:\Temp`), but somehow there's no issue,
2. I've been using the same not-working PC to create the bootable USB drives for many years, but this is the first time I see this `Error: Unable to patch/setup files for boot`. The last time I successfully created the bootable USB drive on this non-working PC was in June this year.

You're also correct that when the Temp folder is not in the standard `C:` drive, there're so many weird issues (mostly with installers).

1. Below are the logs shown in DebugView when Rufus starts on the non-working PC and the working PC:
<details><summary>DebugView - non-working PC</summary>

```
[5660] *** Rufus init *** 
[5660] Could not get actual temp directory from 'C:\PROGRA~3\Temp\': [0x00000001] Incorrect function. 
[5660] Cur dir: 'C:\Users\engr0\Desktop\' 
[5660] App dir: 'C:\Users\engr0\Desktop\' 
[5660] Sys dir: 'C:\WINDOWS\system32' 
[5660] Usr dir: 'C:\Users\engr0' 
[5660] Dat dir: 'C:\Users\engr0\AppData\Local' 
[5660] Tmp dir: 'C:\Users\engr0\Desktop\' 
[5660] Binary executable is signed by 'Akeo Consulting' 
[5660] Will use settings from INI file 
[5660] found locale 'en-US' 
[5660] loc file not found in current directory - embedded one will be used 
[5660] localization: extracted data to 'C:\Users\engr0\Desktop\Ruf3743.tmp' 
[5660] localization: found locale 'en-US' 
[5660] localization: found locale 'ar-SA' 
[5660] localization: found locale 'bg-BG' 
[5660] embedded.loc(834): the version of this translation is older than the base one and may result in some messages not being properly translated.
[5660] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[5660] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[5660] localization: found locale 'zh-CN' 
[5660] localization: found locale 'zh-TW' 
[5660] localization: found locale 'hr-HR' 
[5660] localization: found locale 'cs-CZ' 
[5660] localization: found locale 'da-DK' 
[5660] localization: found locale 'nl-NL' 
[5660] localization: found locale 'fi-FI' 
[5660] localization: found locale 'fr-FR' 
[5660] localization: found locale 'de-DE' 
[5660] localization: found locale 'el-GR' 
[5660] localization: found locale 'he-IL' 
[5660] localization: found locale 'hu-HU' 
[5660] localization: found locale 'id-ID' 
[5660] embedded.loc(5894): the version of this translation is older than the base one and may result in some messages not being properly translated.
[5660] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[5660] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[5660] localization: found locale 'it-IT' 
[5660] localization: found locale 'ja-JP' 
[5660] localization: found locale 'ko-KR' 
[5660] localization: found locale 'lv-LV' 
[5660] localization: found locale 'lt-LT' 
[5660] embedded.loc(7829): the version of this translation is older than the base one and may result in some messages not being properly translated.
[5660] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[5660] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[5660] localization: found locale 'ms-MY' 
[5660] embedded.loc(8209): the version of this translation is older than the base one and may result in some messages not being properly translated.
[5660] If you are the translator, please update your translation with the changes that intervened between v3.22 and v4.5.
[5660] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 
[5660] localization: found locale 'nb-NO' 
[5660] localization: found locale 'fa-IR' 
[5660] localization: found locale 'pl-PL' 
[5660] localization: found locale 'pt-BR' 
[5660] localization: found locale 'pt-PT' 
[5660] localization: found locale 'ro-RO' 
[5660] localization: found locale 'ru-RU' 
[5660] localization: found locale 'sr-RS' 
[5660] localization: found locale 'sk-SK' 
[... 1291 lines cut ...]
Max parts: 128, Start Offset: 17408, Usable = 15376284160 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {1136AE9E-5178-407A-8F93-525676E0E51F}
  Size: 14.3 GB (15374143488 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {92924FB1-D868-48F2-AC76-790E8F8A0A59}
  Size: 1 MB (1064448 bytes)
  Start Sector: 30029706, Attributes: 0x9000000000000000
```
</details>

I understand that it's not logical to spend a lot of effort to fix an issue which affects only a small number of users while there're workarounds (restoring standard Temp folder or using another PC with standard Temp folder).
Will it be better that the moment Rufus detects a symlinked Temp folder, a warning message pops up saying that the process may fails with a 'More details' link to a FAQ or this issue?

As for me, the issue was already considered solved when I successfully created the bootable PC using another PC.
I was just curious why the issue happened today, although there's no issue for the past few years on this same PC, so I didn't close the issue myself. Please close the issue if you don't need additional details.

Thanks for the support and for the good software! 🙏✨