pbatard/rufus #2626: Rufus shows revocation error on fully up-to-date Windows 10 ISO's

thecatontheceiling, 2024-11-24:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus shows the revocation error message when trying to use a fully up-to-date Windows 10 ISO (In this example, one released 5 days ago to MVS users).
Not sure if this is actually a Rufus issue or not, but other people have asked me about why Rufus says the ISO is outdated/has malware even when it is the latest version available and the hashes perfectly match with what is provided at MVS:
[image]

Log
---
```
Rufus x64 v4.6.2208
Windows version: Windows 10 IoT Enterprise S x64 (Build 19044.5011)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:DD24)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x4625C24E
Drive has an unknown Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 28.9 GB (31003246592 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: EFI System Partition (0xef)
  Detected File System: FAT16
  Size: 32 MB (33554432 bytes)
  Start Sector: 60555264, Boot: No
Found 5 additional UEFI revocation filters from remote SBAT
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 10 ISO (Build 19041.5131)
  Size: 6.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_10_consumer_editions_version_22h2_updated_nov_2024_x64_dvd_3eeacab9.iso (6.7 GB)
Note: No Windows User Experience options selected
Warning: '/bootmgr.efi' has been revoked by Windows SVN
Warning: '/efi/boot/bootx64.efi' has been revoked by Windows SVN

Computing hash for 'C:\Users\home-pc\Downloads\en-us_windows_10_consumer_editions_version_22h2_updated_nov_2024_x64_dvd_3eeacab9.iso'...
  MD5:    3bbccf110d9aacf8f658411aecf99fee
  SHA1:   9fa9c88727cfe666367d2db9dd3f3d6422235ca3
  SHA256: fcac487c81c6b15406a78341461d56edb768b00c07cc58e54ee06c48d88ff88a
```

pbatard, 2024-11-24:
Nothing to see here. Microsoft are slow to update their bootloaders, even when they know that they are vulnerable.

As a matter of fact, on a fully up to date system for Secure Boot (i.e. one where you have followed all the steps from [KB5025885](https://support.microsoft.com/en-us/topic/kb5025885-how-to-manage-the-windows-boot-manager-revocations-for-secure-boot-changes-associated-with-cve-2023-24932-41a975df-beb2-40c1-99a3-b3ff139f832d)) **anything** that is signed with the `Microsoft Windows Production PCA 2011` certificate should produce a Secure Boot validation error, and I am not aware of any officially released Windows ISO from Microsoft that does not still use `Microsoft Windows Production PCA 2011` for its bootloaders.

By the looks of it, when Microsoft update the Windows ISO content from an existing release, they only update the content from the wim, and don't bother updating the ISO UEFI bootloaders (as they **should** also do), so the end result is that if the Windows 10 22H2 were vulnerable, which they are, then any updated image based on the Windows 10 22H2 release will still be vulnerable.

So, yeah, if Rufus reports that the UEFI bootloaders from an ISO are vulnerable, then they **are** vulnerable, regardless of whether the people who created the ISO updated it yesterday or not.

closed 2024-11-24 by pbatard

thecatontheceiling, 2024-11-24:
I see, thanks for the information