pbatard/rufus #2627: Rufus fails to format brand new flash drive

Winman486, 2024-11-25:
﻿

Checklist
---------
- [ x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [x ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I just bought a new Kingston DataTraveler Max Type-A 256GB and it will not format at all I ran the bad block test and it came out clean so I am wondering if this flash drive is incompatible?

Log
---
```
<FULL LOG>
```
Rufus x64 v4.6.2208 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.5131)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found 517 revoked UEFI bootloaders from embedded list
Found USB 3.0 device 'WD Elements 2620 USB Device' (1058:2620)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.1) device 'Kingston DataTraveler Max UAS Device' (0951:177F)
1 device found
No volume information for drive 0x86
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {1BC56E55-A123-41A1-A441-36CD01991DDA}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {ED889EC4-3797-4697-BD3B-32E324DF6A9A}
  Size: 238.5 GB (256059408384 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 10 ISO (Build 19041.3803)
  Size: 4.6 GB (Projected)
  Note: File on disk is larger than reported ISO size by 580 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.6 GB)
Note: No Windows User Experience options selected
Warning: '/efi/boot/bootx64.efi' has been revoked by Windows SVN
Warning: '/bootmgr.efi' has been revoked by Windows SVN

Format operation started
Requesting disk access...
Will use 'K:' as volume mountpoint
Opened \\.\PhysicalDrive6 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 488395 (1 block = 512 KB)
Bad Blocks: Using offset 522119 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 238.5 GB)
● Creating UEFI:NTFS Partition (offset: 256059417600, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000015] The device is not ready. - Retrying...
[0x00000015] The device is not ready. - Retrying...
[0x00000015] The device is not ready. - Retrying...
[0x00000015] The device is not ready. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as K: after error

Found USB 3.0 device 'WD Elements 2620 USB Device' (1058:2620)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.1) device 'Kingston DataTraveler Max UAS Device' (0951:177F)
1 device found
No volume information for drive 0x86
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {641B0018-4CEC-411F-AF7D-8BA73220A6C8}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {A4C74BD0-9B23-4359-996E-9983FA5C3F6E}
  Size: 238.5 GB (256058351616 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {1CCE263E-0508-4C9C-BBCD-EF12BDCD67B0}
  Size: 1 MB (1064448 bytes)
  Start Sector: 500116050, Attributes: 0x9000000000000000
Closed Windows format prompt
Found USB 3.0 device 'WD Elements 2620 USB Device' (1058:2620)
Device eliminated because it was detected as a Hard Drive (score 43 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found UAS (USB 3.1) device 'Kingston DataTraveler Max UAS Device' (0951:177F)
1 device found
No volume information for drive 0x86
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {641B0018-4CEC-411F-AF7D-8BA73220A6C8}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {A4C74BD0-9B23-4359-996E-9983FA5C3F6E}
  Size: 238.5 GB (256058351616 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {1CCE263E-0508-4C9C-BBCD-EF12BDCD67B0}
  Size: 1 MB (1064448 bytes)
  Start Sector: 500116050, Attributes: 0x9000000000000000

pbatard, 2024-11-27:
> - [x] I also tried one or more of the following:
>   - [x] Using a different USB drive.
>   - [x] Plugging the USB into a different port.
>   - [x] Running Rufus on a different computer.

Did you **really** see the same issue when running Rufus with a different USB drive and on a different computer?
Can I please see the log(s) from that?

Winman486, 2024-11-27:
I miss checked the box I apologize.

pbatard, 2024-11-27:
What happens if you use `diskpart` to format the drive (as detailed [here](https://www.tomshardware.com/news/format-hard-drive-command-prompt,37632.html) - but you may want to use `format fs=ntfs quick` instead of just `format fs=ntfs`). Do you also get an error?

Winman486, 2024-11-29:
Yes and it tells me there is not enough available space on the drive to
make a partition but its empty and accepts files.

pbatard, 2024-11-29:
As you can understand, if `diskpart` produces the same error when trying to format the drive, then it's clearly not an issue with Rufus.

My guess is that you disabled automounting of drives, which can produce this error, and did not look at the FAQ (that has a dedicated entry for [`Error [0x00000015] The device is not ready`](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Error_0x00000015_The_device_is_not_ready)) which explains what you can try if you run into this error.

But I'm afraid that, outside of that, there's not much I can do to help with pure Windows issues with your environment, so I will close this issue, as it'll be up to you to identify the cause and resolution of your problem (though, if you do find your problem and solution, and it something else than automounting, I'd appreciate if you report here).

closed 2024-11-29 by pbatard