pbatard/rufus #2630: Rufus inconsistent behavior when mountvol /n had been executed

mijioij, 2024-11-26:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
If system is in state where mountvol /e was run, Rufus will succeed creating a drive on both regular usb and UAS (usb mass storage/usb to sata connector/etc...).

If system is in state where mountvol /n is run, Rufus will succeed creating a drive on just regular usb drive but not UAS (usb mass storage/usb to sata connector/etc...).

In the latter case Rufus doesn't choke on the 0x00000015 error and fails later into the process at 0x00000057. I'm not sure if this is intended, but I think the bigger issue here is the inconsistent behavior between regular usb and usb mass storage.

Relevant attempts:
1. I ran mountvol /N in an administrative prompt, opened a new Rufus instance, selected a UAS/USB mass storage drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format and I get the "Format error: Undetermined error while formatting." Log shows 0x00000057 errors. Log reference [1].
2. I ran mountvol /e in an administrative prompt and then opened a new Rufus instance, selected a UAS/USB mass storage drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format then it was able to create the usb and it atleast thinks it succeeded. Log reference [2]. 
3. I ran mountvol /n in an administrative prompt and then opened a new Rufus instance, selected a regular usb drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format, and it was able to format correctly. Log reference [3]. 
4. I then connected the UAS/USB mass storage drive to a different port, started a new Rufus instanced, selected made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format and I get the "Format error: Undetermined error while formatting." Log shows 0x00000057 errors. Log reference [4]

[1]
https://pastebin.com/im51unex
[2]
https://pastebin.com/USd3jNcr
[3]
https://pastebin.com/Hw05h6Tg
[4]
https://pastebin.com/K32xHzKs

pbatard, 2024-11-27:
Please try with the regular non Windows Store version that you can download from https://rufus.ie.

The Windows Store version has various limitations due to how Microsoft restricts what Store applications can do, and that do affect Rufus.

While using the non Store version, try to enable VDS to format the drives (one of the limitations of the Store version is that you cannot use VDS, which is an alternate mode that Rufus can use to format drives). You can toggle VDS with <kbd>Alt</kbd>-<kbd>V</kbd>.

I'm also seeing access denied errors, which may hint that you have an external application interfering with Rufus (because the use of mountvol alone would not be the source of these kind of errors)

mijioij, 2024-11-27:
@pbatard 

Error persists with non-Store version.

5. With mountvol /N run in administrative prompt, opened new instance of non-Store version, selected a UAS/USB mass storage drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format and I get "Format error: Undetermined error while formatting."  Log shows 0x00000057 errors. Log reference [5].
6. With mountvol /e run in administrative prompt, opened new instance of non-Store version downloaded from website, selected a UAS/USB mass storage drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format then it was able to create the usb and it atleast thinks it succeeded. Log reference [6].
7. With mountvol /N run in administrative prompt, opened new instance of non-Store version downloaded from website, selected a UAS/USB mass storage drive, made sure exclusive locking of USB was enabled (Alt+, two times), and checked quick format and I get "Format error: Undetermined error while formatting."  Log shows 0x00000057 errors. Log reference [7].

References:
[5]
https://pastebin.com/pYzGGtJw
[6]
https://pastebin.com/LXJa1H81
[7]
https://pastebin.com/gCWKP5rs

pbatard, 2024-11-27:
What about when enabling VDS as I asked?

mijioij, 2024-11-27:
@pbatard 

I also get the error. Here is log:
https://pastebin.com/SmHxXyPZ

pbatard, 2024-11-27:
Then I have one more test for you: What happens if you use `diskpart` to format the drive (as detailed [here](https://www.tomshardware.com/news/format-hard-drive-command-prompt,37632.html)). Do you also get an error?

mijioij, 2024-11-27:
@pbatard 

For ntfs format step do you want long format like shown in the guide or are you fine with quick format?

pbatard, 2024-11-27:
Quick is fine. That's what Rufus uses.

mijioij, 2024-11-27:
@pbatard 

With mountvol /N, at the format fs=ntfs QUICK step I get the following output:
`
    0 percent completed

DiskPart has encountered an error: The device is not ready.
See the System Event Log for more information.`

While diskpart was open I ran mountvol /e in another window. I then ran the above format command again and got the same error.
I ctrl+c diskpart and then opened it again and ran the above format again and got the same error.
I ctrl+c diskpart and run through the guide again and diskpart was able to successfully format the volume.
I ctrl+c diskpart, run mountvol /N, open diskpart and select this disk and partition and diskpart is able to successfully format.
In same diskpart session as above I ran through guide and I got the above error at the format step.

pbatard, 2024-11-27:
Which, since you get the same error as in Rufus, indicates that it's a pure Windows issue.

Rufus uses the same APIs as `diskpart` for formatting, so the only way for me to attempt to fix this issue is if `diskpart` worked whereas Rufus didn't.

But if both produce the same `device is not ready` error, then it conclusively demonstrates that the issue is internal to Windows, and not something I will be able to address in Rufus (unless I were to attempt to rewrite the Windows formatting API from scratch, which of course, is not a realistic proposition).

The fact that `diskpart` may intermittently work is simply not enough for me to try to work around what is and remains a pure Microsoft formatting API issue (and that might have some components that are tied to your specific environment).

As such, I'm afraid I'm going to have to close this issue, as, clearly, your `device is not ready` errors come from Windows/your environment itself.

closed 2024-11-27 by pbatard