pbatard/rufus #2631: Licencing question about ``mbr_*.h`` files

takusuman, 2024-12-02:
﻿

Checklist
---------
- [X] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I would like to know if the files at ``inc/mbr_*.h`` can have its hexadecimal content copied and re-licenced.
Since it's just dumps from other sources, I do not think a licence is applicable, but I would like to ask you, the developer, before borrowing.
No, I'm not from some faceless company based on China, I'm just trying to make a Go library for reading disk information on UNIX and I would like to know if I can use the hexadecimal dump of bootable partitions as constants on it; Just as most Go libraries, it is under a BSD 3-Clause licence.

Log
---
```
Not applicable 
```

pbatard, 2024-12-02:
> Since it's just dumps from other sources, I do not think a licence is applicable

That's where you are wrong.

It's not because you include a binary that the license goes through the Windows.
The project's license (in this case ms-sys, which is licensed under GPL) does cover the binary data, whether it's an hexdump or something that is compiled from source.

You will therefore **not** be able to relicense these MBR files to the BSD license, unless you can find an alternative source that uses a license compatible with BSD relicensing.

I'm afraid that, whereas one can relicense from BSD 3-Clause to GPL, the reverse is not true.

And, since most of these files originated from a third party project ([ms-sys](https://ms-sys.sourceforge.net/)), and I cannot vouch for the person who licensed these files in the first place, I cannot grand you special rights to relicense them, even if I wanted to.

Unless otherwise specified, everything you can obtain from a project under the GPL, be it source **or binary** falls under the GPL license and can not be relicensed to a different non-compatible license.

closed 2024-12-02 by pbatard

takusuman, 2024-12-02:
G'day.

> unless you can find an alternative source that uses a license compatible with BSD relicensing.

I've already dumped some from my own (such as LILO and bootable NTFS from a Windows 10 Media Creation Tool pendrive), but I will keep searching since it would be less time consuming than remaking all of these from scratch.

> And, since most of these files originated from a third party project ([ms-sys](https://ms-sys.sourceforge.net/)), and I cannot vouch for the person who licensed these files in the first place, I cannot grand you special rights to relicense them, even if I wanted to.

Thanks for the information, I thought it was part of Rufus itself. I will take a look too.

> I'm afraid that, whereas one can relicense from BSD 3-Clause to GPL, the reverse is not true.

Eh, that's true. BSD-esque licences permit relicencing/sublicencing, but not GPL (unless, in the case of relicencing, the community is aware and agrees in majority).

Well, much obliged for the information, and also for Rufus, which I've been using for years.