pbatard/rufus #2635: Running Rufus portable from Powershell CLI, creates a `Rufus.com` (a binary)  file in same directory

eabase, 2024-12-12:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Running Rufus portable from Powershell CLI, creates a `Rufus.com` (a binary)  file in same directory you are located. 

Rufus should not create any *strange* binary files in same directory, when running from shell. 

Log
---
```
n/a
```

pbatard, 2024-12-12:
This is the commandline hogger that is described [here](https://github.com/pbatard/rufus/tree/master/res/hogger).

After having carefully read the notes from the link above, feel free to take it to Microsoft (for not allowing application to be both console and GUI) and Far Manager users if you want to complain.

closed 2024-12-12 by pbatard

eabase, 2024-12-13:
Hi Pete, 
Wow! 🤣 
Crazy. 

```md
Rufus: The Reliable USB Formatting Utility - Commandline hogger

# Description

This little utility is intended to solve the issue of Windows applications not
being able to be BOTH console and GUI [1], leading to all kind of annoyances [2].

The basic problem is as follows:

1. You have an awesome Windows UI application that you want your users to be
   able to start from the commandline (to provide parameters, etc.)
2. If an application is set to operate in UI mode, then when you launch it from
   the command line, you will NOT get a command prompt freeze until it exits, as
   you'd expect from any console app, but instead the prompt will return to the
   user right away.
3. This means that any message that you try to output from your app to the
   console will appear as ugly as:
   C:\Some\Directory> Some Application Message
4. Another unfortunate effect is that, when users exit your app, they might
   continue to wait for the prompt to come back, when it is already available
   whilst polluted by output that looks out of place => Your user experience is
   now subpar...
5. To compensate for this, you might try to be (somewhat) clever, through the
   simulating an <Enter> keypress when your app exit, but lo and behold, soon
   enough you start receiving complaints left and right from Far Manager [3]
   users, due to the fact that this application uses the <Enter> keypress as an
   "I want to launch the currently selected app" event.
6. HEY, MICROSOFT, THIS SUPER-SUCKS!!!!

# Primer

So, how far do we need to go to address this?

1. We'll create a console hogger application, that does just what you'd expect
   a regular commandline app to do (hog the console prompt until the app exits)
   and wait for a signal (Mutex) from the UI app to indicate that it is closed.
2. We'll embed this console hogger as a resource in our app, to be extracted
   and run in the current directory whenever we detect that our UI app has
   been launched from the commandline
3. Because we want this annoyance to have the least impact possible, we'll
   make sure that it is AS SMALL AS POSSIBLE, by writing it in pure assembly
   so that we can compile it with NASM and linking it with WDK, leaving us
   with a 2 KB executable (that will further be compressed to about 1/4th of
   this size through UPX/LZMA).

# Drawbacks

The one annoyance with this workaround is that our 'hogger' will appear in the
command history (doskey). This means that when when a user wants to navigate
back to the command they launched, they need to skip through an extra entry,
which they probably have no idea about. And of course, doskey does not provide
the ability to suppress this last entry.

Oh, and you also need to release the mutex so that you can delete the file 
before you exit, though that's not a big deal...

# Compilation

; From a WDK command prompt

nasm -fwin32 hogger.asm

link /subsystem:console /nodefaultlib /entry:main hogger.obj %SDK_LIB_DEST%\i386\kernel32.lib

# Links

[1] http://blogs.msdn.com/b/oldnewthing/archive/2009/01/01/9259142.aspx
[2] https://github.com/pbatard/rufus/issues/161
[3] http://www.farmanager.com/
```

I guess a work-around would be to create powershell wrapper function that run the program in the *background* and then deletes the file upon exit.

pbatard, 2024-12-13:
`rufus.com` is deleted on program exit. 

There is a small bug (#2622) where this doesn't happen in case you don't launch from the directory where the Rufus executable resides, that will be fixed in the next version.

> so why are we catering to deal with that?

Feel free to take that question with the Far Manager users who happen to be about as annoyed as you are when Rufus does something that annoys them.

eabase, 2024-12-16:
I can confirm that the bug happens, as I launched from random location from powershell, and using the Rufus standalone binary.