pbatard/rufus #2637: Error: (0xC0030570) The file or directory is corrupted and unreadable when attempting to set a Persistent partition size

MSteinbart, 2024-12-15:
Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
## ErrorMessage while formatting USB drive: Error: (0xC0030570) The file or directory is corrupted and unreadable.
I get this error only when the Option `Persistent partition size` is set to any value > 0. If it is 0, no error occurs.

Log
```
Rufus x64 v4.6.2208 (Portable)
Windows version: Windows 10 Pro x64 (Build 19045.5247)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 2.0 device 'usb DISK 3.0 USB Device' (048D:1234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 268 GB, Sector size: 512 bytes
Cylinders: 32635, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x061E7241
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 240.0 GB (257696923648 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 10 GB (10737441792 bytes)
  Start Sector: 503316387, Boot: No
Found 5 additional UEFI revocation filters from remote SBAT
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Reported Grub version: 2.12
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 24.04.1 LTS amd64'
  Size: 5.8 GB (Projected)
  Has a >64 chars filename
  Uses: GRUB2 (2.12)
  Uses: EFI
  Note: This ISO uses symbolic links, which may not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-24.04.1-desktop-amd64.iso (5.8 GB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 240.0 GB)
● Creating Linux Persistence Partition (offset: 257697990144, size: 10 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
327680 possible inodes out of 2621445 blocks (block size = 4096)
131072 blocks (5.0%) reserved for the super user
Creating 80 inode sets: [1 marker = 1.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry

Found USB 2.0 device 'usb DISK 3.0 USB Device' (048D:1234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 268 GB, Sector size: 512 bytes
Cylinders: 32635, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {53183B8C-3A95-496A-8D23-0252D52A912C}
Max parts: 128, Start Offset: 17408, Usable = 268435421696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: (Unrecognized)
  ID: {AA243316-CEB5-4B44-BAE5-8F4C18984B53}
  Size: 240.0 GB (257696923648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Linux Data Partition
  Name: 'Linux Persistence'
  Detected File System: (Unrecognized)
  ID: {29120F59-B704-41E9-A351-EC9098BD2F6B}
  Size: 10 GB (10737441792 bytes)
  Start Sector: 503316387, Attributes: 0x0000000000000000
```

pbatard, 2024-12-15:
Duplicate of #2482.

Your USB drive is defective. If you run a bad blocks check (as was asked from the check list) you should confirm this.

pbatard, 2024-12-17:
No update, so I will assume that the bad blocks check showed your drive was defective and I will close this issue.

closed 2024-12-17 by pbatard