pbatard/rufus #2643: Error: [0xC00305B4] The operation returned because the timeout period expired.

rahullak, 2024-12-20:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Received the below error message when trying to flash Ubuntu iso file onto a USB drive.
[image]

Rufus version: 4.6.2208

Please let me know if any further information is required. Happy to provide!

Log
---
```
Warning: '/EFI/BOOT/BOOTx64.EFI' has been revoked by UEFI DBX
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\rahul\AppData\Local\Rufus\syslinux-6.04\20191223\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Could not get Disk Extents: [0x00000001] Incorrect function.
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.7 GB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x122EE30F
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 3.7 GB (4025450496 bytes)
  Start Sector: 2048, Boot: Yes
Saved C:\Users\rahul\rufus_2.log
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x122EE30F
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 3.7 GB (4025450496 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 489, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x122EE30F
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: (Unrecognized)
  Size: 3.7 GB (4025450496 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2024-12-20:
I'm afraid the issue is with your environment. As suggested by the checklist, try a different drive. a different port, a different computer and you also want to try a bad blocks check.

Also, you truncated the log, so we are missing some data. But I can already state that your issue is 100% environmental and has nothing to do with Rufus.

rahullak, 2024-12-20:
Okay, thank you. I will check the things mentioned in checklist. 

But just to get clarity, is the issue with the drive not having a logical partition? ie. is a logical partition necessary for Rufus to work? The log mentions "No logical drive found" so I wonder if I should just create a logical partition on the drive?

pbatard, 2024-12-20:
Rufus needs to be able to perform various actions to be able to create a drive, including finding a logical drive. If these fail for any reason, you won't just be able to get around them by trying to do the same manually. And since it's only in your specific environment that these common, **standard** operations don't work, whereas they are working for literally tens of million of other Windows users, we can safely conclude that the problem is either with your hardware or with your specific Windows configuration.

closed 2024-12-25 by pbatard