pbatard/rufus #2646: EXT4 support

renatoyamane, 2024-12-25:
Please, add support to EXT4 if you have spare time to do it :-)

I formatted two Flash Drives as EXT3, but one is recognized as EXT2 partition and another as EXT3 partition on my [GL-iNet MT-6000 Router](https://www.gl-inet.com/products/gl-mt6000/) on his native [Network Storage](https://docs.gl-inet.com/router/en/4/interface_guide/network_storage/) feature

[512Gb SanDisk Ultra Luxe Flash Drive](https://www.amazon.co.uk/SanDisk-Ultra-512GB-Flash-Drive/dp/B083ZS6ZR7)

Recognized as EXT2:

Wed Dec 25 10:27:57 2024 kern.notice kernel: [164845.898223] sd 0:0:0:0: [sda] Attached SCSI removable disk
Wed Dec 25 10:27:57 2024 daemon.info samba4-server: io_uring module found, enabling VFS io_uring. (also needs Kernel 5.4+ Support)
Wed Dec 25 10:27:58 2024 kern.info kernel: [164847.175317] EXT4-fs (sda1): **mounting ext2 file system using the ext4 subsystem**
Wed Dec 25 10:27:58 2024 kern.info kernel: [164847.376313] EXT4-fs (sda1): mounted filesystem 206535e8-f529-1b4b-98ed-f9ec87587f8a r/w with ordered data mode. Quota mode: disabled.

[128GB SandDsk Ultra Flair Flash Drive](https://www.amazon.co.uk/gp/product/B015CH1PJU)

Recognized as EXT3:

Tue Dec 24 21:24:02 2024 daemon.err smbd[2545]:   daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
Tue Dec 24 21:24:03 2024 kern.info kernel: [49117.640586] EXT4-fs (sda1): **mounting ext3 file system using the ext4 subsystem**
Tue Dec 24 21:24:04 2024 kern.info kernel: [49117.907970] EXT4-fs (sda1): recovery complete
Tue Dec 24 21:24:04 2024 kern.info kernel: [49117.928506] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
Tue Dec 24 21:24:10 2024 daemon.err nmbd[2546]: [2024/12/24 21:24:10.077506,  0] ../../source3/nmbd/nmbd_become_dmb.c:112(become_domain_master_stage2)

pbatard, 2024-12-25:
Please provide the full log of how you created the drives.

You can access the log by clicking the small logbook icon in Rufus (left of `START`) or by pressing <kbd>Ctrl</kbd>-<kbd>L</kbd>

At this stage, I have no plan to add ext4 support, since the ext formatting is aimed at persistence and using ext4 is an overkill for that. Or you can indicate what feature of ext4, that you don't see in ext3, is something that you see as necessary for you.

I am curious however as to why your first drive is only recognized at ext2, which is why I would very much like to see the complete log of how you created the drive.

renatoyamane, 2024-12-25:
```
Rufus x64 v4.6.2208 (Portable)
Windows version: Windows 11 Home x64 (Build 26100.2605)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.12
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found 517 revoked UEFI bootloaders from embedded list
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk SanDisk 3.2 Gen1 USB Device' (0781:55B1)
2 devices found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 119.2 GB (128042663936 bytes)
  Start Sector: 2048, Boot: Yes
Found 5 additional UEFI revocation filters from remote SBAT
2 devices found
No volume information for drive 0x83
Disk type: Removable, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60867, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0E3C1A71
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: (Unrecognized)
  Size: 466.3 GB (500647827456 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive3 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 466.3 GB)
Waiting for logical drive to reappear...
Notice: Using physical device to access partition data
Formatting (ext3)...
7639279 possible inodes out of 122228473 blocks (block size = 4096)
6111423 blocks (5.0%) reserved for the super user
Creating 3731 inode sets: [1 marker = 46.6 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 131072 journal blocks: [1 marker = 1638.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Opened \\.\PhysicalDrive3 for shared write access
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{2dd2fa70-c231-11ef-8716-e8bfb80d1dde}\
Successfully remounted \\?\Volume{2dd2fa70-c231-11ef-8716-e8bfb80d1dde}\ as G:

Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk SanDisk 3.2 Gen1 USB Device' (0781:55B1)
2 devices found
No volume information for drive 0x83
Disk type: Removable, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60867, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 466.3 GB (500647827456 bytes)
  Start Sector: 2048, Boot: No
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0764)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk SanDisk 3.2 Gen1 USB Device' (0781:55B1)
2 devices found
No volume information for drive 0x83
Disk type: Removable, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60867, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 466.3 GB (500647827456 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2025-01-03:
Considering that nobody else is reporting an issue with `ext3` being recognized as `ext2` and that you seem to be using a proprietary version of Linux (as opposed to a mainstream distro), I'm going to dismiss the `ext3` being recognized as `ext2` as a non Rufus issue, unless you test that drive on a recent Ubuntu/Debian/Mint (or whatever well known mainstream popular release you like) machine and happen to encounter the same error.

The only issues I can address are the ones that deal with generic hardware (i.e., in your case that would be: _"Any ~512 GB drive formatted as `ext3` gets mounted as `ext2` in every recent Linux distro I have tested it with"_), because there are just too many quirks that can happen when specific hardware X is used with specific hardware and proprietary software Y, and, as you might expect, I neither have the resources nor the time to go around and buy specific hardware X and Y _in the hope_ that I will be able to replicate the issue.

Therefore, unless someone reports `ext3` drives created by Rufus being mounted as `ext2` with generic Linux distros, I'm simply going to have to dismiss this issue as environmental.

closed 2025-01-03 by pbatard