pbatard/rufus #2649: Include Syslinux 6.04/20191223

RustoMCSpit, 2025-01-01:
Issue description
-----------------
<Please describe your issue here>

I know Syslinux and GRUB are pains, I read the Security information. My question is simple, could you not just include the new Syslinux into the application itself? I mean, why break them up?

pbatard, 2025-01-01:
Because, due to the **super slow** times between official releases of GRUB and Syslinux, distros create their own "updated" version of GRUB/Syslinux, that oftentimes breaks boot when used with the official release files (from the top of my head, Mint, Fedora and Ubuntu did that, and I'm probably forgetting some more).

Which means that there's just none one GRUB 2.06 that works with everything and that I could include in Rufus, but multiple versions, that need to be detected and sometimes rebuilt in a hurry by yours truly, with the various breaking patches that this and that distro has added, so that boot will actually works.

Also, using a download ensures that:
1. I don't have to go through a release of Rufus the minute I discover GRUB/Syslinux breakage that needs to be fixed with a custom new bootloader (because the current GRUB/Syslinux detection system tries to look for a custom version first, and only downloads regular if it can't see one on our servers, meaning that fixing a boot breakage can be sorted **without having to ask users to update any software on their side**)
2. People using older versions of Rufus benefit from these updates.

So, from where I stand, juggling something like 30 versions of bootloaders in Rufus and having to go through a hurried release the day a 31st one is found to be required while forcing every user of Rufus to update at once if they want to benefit from the fix does not make a shred of sense, when ALL of these issues can be alleviated by using online downloads (please understand that if data is provided online, it's usually because the developer has **very carefully** considered the benefits and drawbacks of online vs embedded)... at the small hindrance of annoying a very fringeset of users who feel compelled to use Rufus offline (whereas Rufus is **not** designed to be used offline).

closed 2025-01-01 by pbatard

RustoMCSpit, 2025-01-01:
alright cool, thanks! you can add this answer as a dropdown to those curious in the security wiki