pbatard/rufus #2650: Feature Request: Add "disable device encryption" to pre-installation of Windows 11 ISO, especially 24H2

Meyers07, 2025-01-02:
So after seeing issue 2244:
https://github.com/pbatard/rufus/issues/2244

I managed to create a Windows 11 24H2 ISO with requirement for 4gb, secure boot, and uefi removed, requirement for online microsoft account removed, and bitlocker device encryption (which was said to be made automatically turned on for all editions since 24H2 fresh installation) disabled.

However a separate "device encryption" is still partially turned on, as seen in the pictures below (i haven't take picture of the device given i take it for physical repairs for hardware issues unrelated, specifically a completely malfunctioning wi-fi card, but its taken from a reddit post with same issues)

i can simply turn off device encryption but that means decrypting, which would be longer to process on a filled system (my system currently is empty apart from Windows 11 24H2 installation and i plan to clean reinstall it after the previously-mentioned hardware issues are fixed).
here's the question: can rufus be configured to turn off the other device encryption setting?

[image]

[image]

closed 2025-01-02 by pbatard

Meyers07, 2025-02-27:
Not meant to reopen but after i re setup a usb boot and reinstall my system (unrelated to this issue) this issue doesn't appear in one of the newer version of Windows 11 24H2 ISO (the January 2025 update), on the newer ISOs the device encryption setting is never turned on following the disable bitlocker automatic device encryption setting. Which works as intended. 
With detailed image identification file of the newer ISO i mentioned:
CRC-32:	3e3c255a
MD5:	d77a8d85da7e23730b04354e6a47d6ee
SHA-1:	59a084a9425139aaae40370b04bf5b69ccd31cef
SHA-256:	622eed9bc88fa91623282aeb197dcdd4227f4aea24dea051ff099d1e306ddbeb

I guess the bug lies on the RTM* version of Windows 11 24H2 ISO released in September 2024 where it didn't respond to that command.
*detailed image identification file
CRC-32:	a0d29b18
MD5:	bb2acb101d4a360c71530e4be00ec43c
SHA-1:	a81079ba4b9a9ce29a7a2a20cd34266ac21becce
SHA-256:	b56b911bf18a2ceaeb3904d87e7c770bdf92d3099599d61ac2497b91bf190b11
SHA-512:	4292beb7cd8e3aa75a3e07e3426d49def735cca6b8914456530f9541bf02bd913710117d5bd2492d2002774a13916b951b8ab7012bee5322decacf3c3224a0a7

pbatard, 2025-02-27:
Thanks for the note. It is always possible (and judging by the number of redo/holds that Microsoft issued on feature updates) that Microsoft screws something up in a specific image indeed, so I'll take your word that some of the MSDN-only accessible monthly ISOs might have botched the bitlocker encryption settings at some stage.