pbatard/rufus #2653: OLD issue I can't resolve

dustinoff, 2025-01-07:
﻿

Checklist
---------
- [x] I have been made aware that if my problem can be summarized as _"I've created or tried to create a media using Rufus, and ..."_, and I am not including a log, **this issue will be summarily closed**.
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am missing something. I downloaded W11 24H2 from MS. I have Rufus 4.6p. I open Rufus select my device. Select the ISO image. Select the partition scheme and file system and Rufus scans the image. It then shows me image options with 2 choices Standard windows installation and Windows to go. There is no extended option. I click Start and get the warning about formatting. What did I do wrong to not get the extended option?

Log
No log cause I haven't run it yet.
<FULL LOG>
```

pbatard, 2025-01-07:
> No log cause I haven't run it yet.

Then you get no support here. It's that simple. You **do** get a log when you select an ISO, that would tell us what your issue is. But you **chose** to ignore what was specifically asked of you to provide, so all I will say is good luck fixing your issue.

closed 2025-01-07 by pbatard

pbatard, 2025-01-07:
> maybe you can tell me where that log file is

From the check list item you ticked:

> - [x] I clicked the 'Log' button (🗒️) or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.

In other words, you explicitly told everyone, when opening this issue, that you did something (copy/paste the log) which you did not do at all, and furthermore, didn't even bother reading the check list at all, as it would have answered the question you had above.

Please don't bother posting a log here. I have zero time for people who clearly think reading and following instructions is beneath them, and I am pretty sure, judging from your prior behaviour, that if that is your approach to asking for help, then trying to help you will be a complete waste of time.