pbatard/rufus #2658: Revoked UEFI bootloader detected in latest version of Windows 11 23h2 ISO with Rufus 4.6

ayaya2036, 2025-01-17:
I download latest version of Windows 23H2 (22631.4751) from UUP dump and this error pop up when clicking start on Rufus 4.6.

[image]

I try to use an older version of Rufus (4.5) and the error doesn't pop up.
I tried some other version of 23H2 ISO (22631.4602) and (22631.4169) and the error still pop up in Rufus 4.6.
Does Rufus 4.6 flags all 23H2 ISO containing revoked UEFI bootloader now?

pbatard, 2025-01-17:
Next time, please check the FAQ as you were asked:

pbatard, 2025-01-17:
The reason why older versions of Rufus might not produce a warnings is that they do not have the updated checks for revocations that newer versions of Windows use.

closed 2025-01-17 by pbatard