pbatard/rufus #2662: stop code: driver irql not less or equal

StefanStHH, 2025-01-22:
I received a bluescreen when trying to install windows 11 24h2 created with rufus 4.6.2208 
on an sony vaio vpcsa laptop. the error code says:

stop code: driver irql not less or equal
what failed: Wdf01000.sys

I would appreaciate any help! Thank you in advance!

pbatard, 2025-01-22:
I'm afraid this is a pure Windows issue that has nothing to do with using Rufus (Rufus does **NOT** modify any of the Windows binaries, even if you apply the TPM bypass).

You will need to post this question in a Windows user forum, which this issue tracker isn't.

closed 2025-01-22 by pbatard

epman100, 2025-02-11:
actually this is more of a Windows compatibility issue as the VPCSA series of Sony laptops are NOT fully compatible with either Windows 10 or Windows 11 and may crash into blue screen because of having a "hybrid" or a mixture of switchable onboard graphics/video hardware, according to these Sony support pages like [this one](https://www.sony-asia.com/microsite/support/win10/en/windows10_7.html) & [this one](https://www.sony.co.uk/electronics/support/articles/00121830).  these VPCSA laptops from Sony were only designed up to running Win7.

OP should **avoid** attempting to install either Win10 or Win11 on such old Sony laptops **at all costs**.

StefanStHH, 2025-02-11:
The issue was windows 11 24h2 January Update. Laptop works fine now with installing windows 10 22h2 (necessary because of mbr requirement) and upgrading to windows 11 23h2 with all updates up to today. The dual graphics solution seems to be no problem, windows uses intel as standard and the Radeon driver program can be used to start the and gpu for a specific program (e.g. a game on steam) The hardware switch to go from stamina to power mode does not interfere with windows, but seems to tell the motherboard on bios level to either run the cpu full power or throttle it up and down depending on temp. All buttons are working as intended after installing the proper original drivers in campatibility mode.

Hope this helps! Have a good one!

Stefan Stiegeler
Mobile: +49 151 1744 6094

StefanStHH, 2025-02-11:
Correction: windows 10 needed to be installed first due to some of the hardware was not supported by windows 11. so n windows 10 install worked and drivers install was possible after installing. Then upgrade to windows 11 with all drivers already working.

Stefan Stiegeler
Mobile: +49 151 1744 6094