pbatard/rufus #2669: Win11 2H24 won't install from USB when booting from Clover 5141

KaptenLutman, 2025-01-24:
I have an old, non-UEFI HP Z800 which I'm booting using Clover (version 5141, boot file from version 3911 since 5141 won't boot). I installed a new NVMe SSD and tried to use Rufus 4.6 to freshly install Win 2H24 on the new SSD. I can boot the Rufus-created USB from Clover (using UEFI shell command fs1:\EFI\boot\bootx64.efi) and start the installation, but at the end I get an error message "Windows 11 failed to install". I tried with several USB sticks but I always get this error.

Instead I used Rufus 4.6 (checking the USB drives box) to create the installation disk on a SATA6 SSD in a USB3 case. After completion, I moved the SATA SSD to the Z800, booted it via Clover and, voila, installation worked!

JonnyTech, 2025-01-24:
> non-UEFI

Windows 11 requires UEFI system

https://www.microsoft.com/en-us/windows/windows-11-specifications

Your hack may well bypass the Windows installer trying to update the UEFI bootloader at the end of the installation, but you may experience issues with further updates.

Anyway, this is not a Rufus issue and your edge-case requirements mean that you are on your own for such an advanced and rare situation.

pbatard, 2025-01-25:
Agreed. I will therefore close this issue.

closed 2025-01-25 by pbatard