pbatard/rufus #2671: just tried to use rufus4.6 to get windows 11 and it broke my new usb drive...

YellOhGinner, 2025-01-25:
it was a 128gb usb stick, it was brand new, barely used. i thought about getting the windows 11 today and i was following simple steps from a youtube video. after pressing "start" on the rufus app and after about 20 - 30 mins of waiting, i noticed that it was about to be completed so i stayed and watched. right after the loading thing completed, the windows just closed and rufus was staring at me without any further instructions and the difference i saw right away was my pendrive now not showing up anywhere., not in the app, not in the disk management thing.  besides the notfication about my pendrive not being recognised when i plug it in to check, theres nothing. 

one more information: when i pulled out my pendrive, it was fricking hot asf. Rufus cooked it. i dk whats the next step, ive looked up videos that say that following those steps fixes the issue but none of them worked as none of thier problems were created using rufus. i would love some help trying to get beck my pendrive to a working state. ty.

YellOhGinner, 2025-01-25:
sorry if theres not much information provided here, im quite new to forums and really basic at techs. id be happy to send logs and stuff. i cant read them.

JonnyTech, 2025-01-25:
> when i pulled out my pendrive, it was fricking hot asf

The Windows ISO is over 5.5Gb which is written to the drive. Some cheap / fake / sub-standard drives cannot take the sustained write operation. It is not a Rufus or a Windows issue, but hardware. If the drive cannot dissipate the heat or otherwise limit it's operation then it will cook. I suspect that there is no chance of recovery, even after it cools down. Send it back to your supplier and get a RMA for it.

pbatard, 2025-01-25:
Yes, I'm afraid that drive failures do happen regardless of the application you use (since flash drives are a lot more unreliable than people think), and it's quite normal for **properly built drives** to be able to sustain high temperatures when writing a lot of data at a fast sequential rate, which Rufus might do.

I'm afraid that whatever hardware issue you experience had nothing to do with Rufus (your drive would have failed with any other application or even Windows explorer if you copied the same amount of data), so there isn't any help I can offer here.

Please also see: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive

closed 2025-01-25 by pbatard

YellOhGinner, 2025-01-25:
> The Windows ISO is over 5.5Gb which is written to the drive. Some cheap / fake / sub-standard drives cannot take the sustained write operation. It is not a Rufus or a Windows issue, but hardware. If [...]

that makes alot of sense. thank you for taking your time to help :)

YellOhGinner, 2025-01-25:
> Yes, I'm afraid that drive failures do happen regardless of the application you use (since flash drives are a lot more unreliable than people think), and it's quite normal for **properly built [...]

im convinved that it was the heat now. it was just that ive had older usb sticks with linux mint iso inside using ventoy and never faced such issue. its my first. thank you!

JonnyTech, 2025-01-25:
> using ventoy

FYI reading is much less stressful than writing and generates practically no heat. Booting from a drive (using ventoy, etc) should be no issue.

Plus, a Mint ISO is less than half the size of Windows. Doing the initial write to a drive would generate significantly less heat.