pbatard/rufus #2672: Template missing for new issue

JonnyTech, 2025-01-25:
@pbatard hit the new issue button on Github...

[image]

pbatard, 2025-01-25:
Ah what the £$%^&*! Looks like GitHub completely reworked their issue template handling, without providing any hint that they were introduced breaking changes (whereas it would have been super easy for them to look at all the repos that used `.github/issue_template.md` and e-mail a notice).

Thanks for the heads up. Let me see if I can fix this.

closed 2025-01-25 by pbatard

pbatard, 2025-01-25:
Sheesh, what a piece of crap this new GitHub template handling is. Should hopefully be fixed now.

JonnyTech, 2025-01-25:
Confirmed as fixed - well done!

commit fe795bfa (2025-01-25): [misc] Fix issue template no longer being processed

* Since GitHub broke existing paths and did not bother to alert people.
* Closes #2672.
commit 23e092ac (2025-01-25): [misc] Fix issue template no longer being processed

* Since GitHub broke existing paths and did not bother to alert people.
* Closes #2672.
commit d83b8ead (2025-01-25): [misc] Fix issue template no longer being processed

* Since GitHub broke existing paths and did not bother to alert people.
* Closes #2672.
commit ba49668a (2025-01-25): [misc] Fix issue template no longer being processed

* Since GitHub broke existing paths and did not bother to alert people.
* Closes #2672.